nvs-vocabs/L22 #1: Request for new L22: Hyperbenthic sledge

Daphnisd, 2018-11-30:
Label: “Hyperbenthic sledge”

Proposed definition “A hyperbenthic sledge is used for the sampling of hyperbenthos. The sledge is divided into two compartments one above the other, each accommodated with a hyperbenthos net. In this way, a distinction can be made between hyperbenthos close to the seafloor and benthos occurring a little higher. There is some space between the bottom and the nets, so the nets float in the water column and don’t catch any sediment.”

https://github.com/iobis/vocab/issues/5

louatbodc, 2018-12-03:
Hi Daphnis

Are you looking to create a generic term for a hyperbenthic sledge or are you wanting a term for the one at VLIZ?

Many thanks
Lou

louatbodc, 2018-12-03:
Hi Daphnis

Reading around, I would prefer to use a more generic description than the one provided by VLIZ. It appears as through there are many slight variations but a typical design and method of deployment. I propose the following:

Unspecified hyperbenthic sledge
--

A benthic sledge used for sampling benthic organisms that live just above the sediment (hyperbenthos). They are  typically designed to measure the lower 1 m of the water column and normally consist of a heavy, metal frame with two or more nets on top of each other that are 0.5 or 1 mm mesh size. They are equipped with an opening-closing device to prevent contamination with organisms from the upper water column and are towed from a large vessel such as a research ship. Most variations incorporate some space between the bottom and the nets, so the nets float in the water column and don’t catch any sediment.
--

Daphnisd, 2018-12-04:
I think this will be fine.
Thnx Lou!

closed 2018-12-05 by louatbodc

gwemon, 2018-12-06:
Link to new term is http://vocab.nerc.ac.uk/collection/L22/current/NETT0189/