nvs-vocabs/L22 #20: NTR:

mcuthill, 2020-02-28:
Need a new term for the RDI Workhorse Monitor 1200kHz (terms exist for the 300kHz and 600kHz):

## Device make: Teledyne RDI
## Device model: 1200 kHz Workhorse Monitor (WHMZ1200)
## Model version: I-UG49 (maybe? The full model # is WHMZ1200-I-UG49 according to our documentation)
## Model alternative name (if applicable): RDI Workhorse-1200 DR-ADCP
## Qualification (if applicable): direct-reading
## Instrument type: ADCP
## Does the tool describe a series of instruments? n/a
## Reference: n/a
## Short name: RDI 1200kHz Workhorse Monitor ADCP
## Documentation
http://www.teledynemarine.com/workhorse-monitor-adcp for general info, see attachment for manual (covers all 3 frequencies of this instrument model):
[RDI ADCP WorkHorse Technical Manual- April 2005.pdf](https://github.com/nvs-vocabs/L22/files/4269551/RDI.ADCP.WorkHorse.Technical.Manual-.April.2005.pdf)

louatbodc, 2020-03-03:
Many thanks for your request. We will be dealing with your request shortly.

Many thanks
Lou

louatbodc, 2020-03-06:
I believe UG refers to upgrades which are optional configurations available with the standard monitor WH1200. I have created a new term for the workhorse monitor 1200 kHz that includes descriptions of possible upgrades.

Teledyne RDI Workhorse Monitor 1200 kHz (direct-reading) ADCP (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1406/)

Please note is can take up to 24 hrs for the term to appear on our server and websites.

Cheers
Lou

closed 2020-03-06 by louatbodc