nvs-vocabs/L22 #23: NTR:

mcuthill, 2020-03-11:
## Device make
BioSonics, Inc

## Device model
Digital Scientific Echosounder

## Model version
DT-Xu

## Model alternative name (if applicable)

## Qualification (if applicable)

## Instrument type
Echosounder 

## Does the tool describe a series of instruments? 

## Reference

## Short name
Biosonics Echosounder DT-Xu

## Documentation
No manual or spec sheet is available from the manufacturer for this device. We do have a user guide for the proprietary software it shipped with, which I'm attaching; it discusses the instrument hardware itself in the first couple of chapters, which may be sufficient. 
[manVisAcq.pdf](https://github.com/nvs-vocabs/L22/files/4320862/manVisAcq.pdf)

## Mapping to external terminologies (optional)

## ORCID (optional)

vpaba, 2020-03-12:
Hello, 
Thanks for your submission. I will be dealing with this request early next week.
Kind regards,
Violetta Paba - BODC

vpaba, 2020-03-19:
Hello,

Thanks for your patience. The L22 requested has now been created:

TOOL1408 - BioSonics DT-X

Please note that it may take 24h for this new entry to appear on the NVS.
Any issues with this new vocabulary concept, please let me know.

Kind regards,

Violetta Paba - BODC

closed 2020-03-19 by vpaba