nvs-vocabs/L22 #25: Duplicate terms found

mcuthill, 2020-04-04:
Searching for a term for an Aanderaa Oxygen Optode 4330, I found 3 entries:

http://vocab.nerc.ac.uk/collection/L22/current/TOOL0826/
http://vocab.nerc.ac.uk/collection/L22/current/TOOL1247/1/
http://vocab.nerc.ac.uk/collection/L22/current/TOOL1248/

Please let me know which I should use!
Thanks

roy-lowry, 2020-04-04:
It depends upon how detailed the information you have about your 4330. If you know it was a 4330 with a normal foil fitted then use 1247. If you know it was a 4330F (the fast response version of the sensor fitted with a different internal foil) then use 1248. If you are not sure which foil was fitted to the sensor then use 826. If at all possible use 1247 or 1248.

Notes to BODC DPT team

I noticed a typo in the Alt label field of 826 (4430 for 4330).  Thought I'd respond to help ease loads in the current situation. Not sure what request management is required so could somebody do that for me please?

gwemon, 2020-04-04:
Thanks @roy-lowry . Correction to Alt label made. It'll be live later on today.

mcuthill, 2020-04-09:
Thanks @roy-lowry!

closed 2020-04-09 by mcuthill