nvs-vocabs/L22 #26: Request for feedback on new L22 terms

danibodc, 2020-04-06:
We have had a request for some new L22 terms relating to colonisation substrates (e.g. settlement plates), deployed to measure the settlement/colonisation of marine organisms. 

Examples of L22 preferred labels:

· Lexan polycarbonate colonisation substrate

· Unknown polyvinyl chloride (PVC) pipe colonisation substrate

· Scotch-Brite pad bound to polyvinyl chloride (PVC) colonisation substrate

· Unknown grooved polyvinyl chloride (PVC) colonisation substrate

· Unknown ceramic tile colonisation substrate

We would appreciate any feedback on the above terms.

roy-lowry, 2020-04-06:
I seem to remember a discussion on these concepts during a VMG meeting and from what I remember these seem to be what I would expect in L22 linked to a colonisation substrate L05 entry. I think this was being set up by Lou, but not sure how far it got. 

I don't like the word 'unknown' in concept labels- I'd use the description to explain the generic nature of the concept. I also don't like abbreviations in labels - the purpose of CODALT is to expose abbreviations to searches. For example I'd have:

Polyvinyl chloride pipe colonisation substrate
PVC pipe substrate
Any object placed in the marine environment for attachment of organisms that is a hollow cylinder with open ends made from the plastic polyvinyl chloride (PVC).

for codnam, codalt and description.

danibodc, 2020-04-07:
We have now finalised the following terms, which will appear on the NVS overnight:

Custom-made Lexan plastic 6-plate settlement panel (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1413/)
Custom-made grooved polyvinyl chloride half-pipe (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1414/)
Custom-made ceramic tile settlement plate (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1415/)
Custom-made 3M Scotch-Brite pad on polyvinyl chloride settlement plate (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1416/)
Custom-made grooved polyvinyl chloride settlement plate (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1417/)

These terms have been linked to the following L05 term:
colonisation substrata
(http://vocab.nerc.ac.uk/collection/L05/current/392/)

And the following L35 term:
Custom-made
(http://vocab.nerc.ac.uk/collection/L35/current/MAN0069/)

closed 2020-04-07 by danibodc