nvs-vocabs/L22 #28: NTR: multiple terms

mcuthill, 2020-04-14:
Bulk Excel sheet:
[ONC_batch_1_L22_submissions.xlsx](https://github.com/nvs-vocabs/L22/files/4477673/ONC_batch_1_L22_submissions.xlsx)

Supporting documentation where no web link available:
[ASL WCP manual CM-100-WCP-01-R01.pdf](https://github.com/nvs-vocabs/L22/files/4477685/ASL.WCP.manual.CM-100-WCP-01-R01.pdf)
[Scientific Fishery Systems SciFish2100 spec sheet.pdf](https://github.com/nvs-vocabs/L22/files/4477708/Scientific.Fishery.Systems.SciFish2100.spec.sheet.pdf)
[RM_YOUNG ANEMOMETER_041065_manual.PDF](https://github.com/nvs-vocabs/L22/files/4477713/RM_YOUNG.ANEMOMETER_041065_manual.PDF)
[Edgetech_4200_series_side_scan_sonar_brochure.pdf](https://github.com/nvs-vocabs/L22/files/4477717/Edgetech_4200_series_side_scan_sonar_brochure.pdf)

louatbodc, 2020-05-19:
The following terms were published:

Aanderaa Oxygen Optode 3930 oxygen sensor (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1421/)
AML Oceanographic Minos X CTD (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1422/)
EdgeTech 4200 Side scan Sonar (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1423/)
HS Engineers series-2001 current meter (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1424/)
Nortek Vectrino 3D Acoustic Velocimeter (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1425/)
Pro-Oceanus HGTD-Pro {Hurricane} dissolved gas sensor (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1426/)
RM Young 04106 Wind Monitor JR (marine) anemometer (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1428/)
RM Young 05103 Wind Monitor 10A anemometer (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1429/)
RBR Virtuoso 3 (submersible) data logger (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1430/)
RBR Quartz 3 BPR bottom pressure recorder (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1431/)
Rotronic AG HC-S3 XT temperature and humidity probe (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1432/)
Trimble NetR8 GNSS reference station (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1433/)
Trimble NetR9 GNSS reference station (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1434/)
Trimble NetRS GPS reference station (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1435/)
Scientific Fishery Systems SciFish 2100 A (Split-beam) echosounder (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1442/)
Scientific Fishery Systems SciFish 2100 B (Broadband) echosounder (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1443/)
Scientific Fishery Systems SciFish 2100 C (Multi-frequency) echosounder (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1444/)
Scientific Fishery Systems SciFish 2100 D (Combined) echosounder (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1445/)
Topcon Positioning Systems NET-G3A GNSS Receiver (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1446/)
Turner Designs Cyclops-7F fluorometer (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1447/)
Vaisala CS106 barometer (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1436/)
Valeport miniCTD DR CTD (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1437/)
WET Labs ECO-FL (S) {FLS} fluorometer (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1438/)
WET Labs ECO-FL (RT)D fluorometer (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1439/)
WET Labs ECO-FL (B) {FLB} fluorometer (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1440/)
WET Labs ECO-FL (SB) {FLSB} fluorometer (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1441/)

louatbodc, 2020-05-19:
Within this batch there was a request for a new code for 'ASL Environmental Water Column Profiler {WCP} (acoustic) water column profiler'
Our research points to the water column profiler (WCP) being the same as an Zooplankton profiler (ZAP) and suggest using the term TOOL1420.

In addition, one request for a WETLabs ECO-FL(RT) real-time fluorometer already exists (code TOOL1392)

mcuthill, 2020-05-22:
Thank you!
Unfortunately your reward will be another big batch of requests coming your way...

closed 2020-05-22 by mcuthill