nvs-vocabs/L22 #34: Bulk new term request, round 2

mcuthill, 2020-06-05:
Attaching Excel sheet and documentation where unavailable online.
Side note: I like the new request template! It makes it much easier to pin down the granularity. Although the addition of the Modification field just set my 'map all our instruments' project back a few weeks, dang it...

[ONC_batch_2_submissions_v2.2.xlsx](https://github.com/nvs-vocabs/L22/files/4738755/ONC_batch_2_submissions_v2.2.xlsx)

Documentation:
[AMT MOS O2-2000m.pdf](https://github.com/nvs-vocabs/L22/files/4738724/AMT.MOS.O2-2000m.pdf)
[GEOSense_BH_1_Version_5_Manual_Upgrade_II-1.pdf](https://github.com/nvs-vocabs/L22/files/4738726/GEOSense_BH_1_Version_5_Manual_Upgrade_II-1.pdf)
[Imagenex_Delta_T_837__Specs_rev4.pdf](https://github.com/nvs-vocabs/L22/files/4738733/Imagenex_Delta_T_837__Specs_rev4.pdf)
[RBR virtuoso datasheet.pdf](https://github.com/nvs-vocabs/L22/files/4738749/RBR.virtuoso.datasheet.pdf)
[MBARI-ISUSXv2-Operation_Manual.pdf](https://github.com/nvs-vocabs/L22/files/4738752/MBARI-ISUSXv2-Operation_Manual.pdf)
[Sunburst_Sami2CO2_07jul2011.pdf](https://github.com/nvs-vocabs/L22/files/4738753/Sunburst_Sami2CO2_07jul2011.pdf)
[Turo PT100 to RS232 Manual.pdf](https://github.com/nvs-vocabs/L22/files/4738754/Turo.PT100.to.RS232.Manual.pdf)

louatbodc, 2020-08-03:
The following terms have been published. Please note, it may take up to 24 hours for terms to appear on our web services:

AMT Analysenmesstechnik MoS oxygen sensor (TOOL1509)
Campbell Scientific CR800 data logger (TOOL1510)
Davis Instruments Vantage Pro2 Plus 6162 (wireless) weather station (TOOL1511)
Davis Instruments Vantage Pro2 Plus 6162C (cabled) weather station (TOOL1512)
Geosense BH-1 Version 5 corehole seismometer (TOOL1513)
Hemisphere GNSS V104s GPS compass (TOOL1514)
High Tech Inc HTI 96-Min hydrophone series (TOOL1497)
High Tech Inc HTI 99-HF hydrophone (TOOL1498)
High Tech Inc HTI 99-UHF hydrophone (TOOL1499)
Imagenex Delta T 837 (profiling) multibeam echosounder (TOOL1500)
Imagenex Delta T 837A 3000 m (profiling) multibeam echosounder (TOOL1501)
Kipp and Zonen CMP21 pyranometer (TOOL1502)
Ocean Sonics icListen AF hydrophone (TOOL1504)
Ocean Sonics icListen LF hydrophone (TOOL1505)
RBR Virtuoso (single channel) data logger (TOOL1506)
Satlantic MBARI-ISUS-X V2 ultraviolet spectrophotometer (TOOL1507)
Sea-Bird SBE 9plus CTD (TOOL1508)
Sea-Bird Slocum Glider Payload {GPCTD} CTD (TOOL1492)
Sunburst Sensors Submersible Autonomous Moored Instrument 2 {SAMI2-pH} pH sensor (TOOL1493)
Valeport miniSVP sound velocity profiler (TOOL1494)
Valeport miniSVS sound velocity sensor (TOOL1495)
Turo Technology PT100 temperature sensor (TOOL1503)

The following terms existed:
TOOL0834 WETLabs ac-s in-situ spectrophotometer

Kind regards
Lou

mcuthill, 2020-08-05:
@louatbodc - Thank you for the new terms! Regarding TOOL0834 with the existing term, our resident point-person for these devices found that the WETLabs ac-s in-situ spectrophotometer isn't completely identical to the Sea-Bird ac-s Spectral Absorption and Attenuation Sensor. According to him, the two instruments have slightly different wavelength ranges: the Sea-Bird has roughly 80 wavelength outputs and Outputs from 400 to 700 nm, while the Wet Labs one had 81 different wavelengths ranging from 400 to 756 nm.

This may be too minor a difference to warrant a different term, but we wanted to make sure it was taken into account when making the decision. Can you comment on this?

louatbodc, 2020-08-06:
Hi @mcuthill 

Thanks for this information. We'll look into it for you

Cheers
Lou

roswri, 2020-09-14:
Hi @mcuthill,

I have checked the specification sheets by Wetlabs (https://www.bodc.ac.uk/data/documents/nodb/pdf/Wetlabs_ac-s_spectrophotometer_brochure.pdf) and Seabird (https://www.seabird.com/asset-get.download-en.jsa?id=54627862140) and they both quote exactly the same numbers and from what I can tell seem to be exactly the same instrument.
The only reference to the ACS having had 81 different wavelengths ranging from 400 to 756 nm: https://seabass.gsfc.nasa.gov/archive/URI/Mouw/LakeSuperior/LCCMR201507/documents/WetlabsACS_Mouw_2015.pdf which references the 2011 document, I think the one we have is from 2013, so I'm not sure if the specifications changed. Does your colleague have access to any official documentation that confirms the differences between the instruments? 

Many thanks,
Roseanna

JordanAthertonNOC, 2023-02-07:
Hi @mcuthill 

We are doing a tidy up of old tickets - I have also reviewed the above and the only other resource I can find referencing the 400 - 756 nm range is the following paper from 2021: https://onlinelibrary.wiley.com/doi/full/10.1002/rra.3771 
 
However the official Sea-Bird/WET Labs documentation still only references a range of 400 - 730 nm. I am therefore closing this ticket, feel free to re-open it if further information comes to light shedding light on the discrepancy.

Best regards

Jordan