nvs-vocabs/L22 #36: New Kongsberg HydroC CO2 term published [BODCNVS-555]

louatbodc, 2020-06-22:
A new term was published via an email request.

Kongsberg Maritime CONTROS HydroC (underwater) CO2 sensor (TOOL1483)

Please note, it may take up to 24 hours for new publications to appear on our web services

closed 2020-06-22 by louatbodc

hahn-johannes, 2020-08-12:
Adequately, I would be good to register the "Kongsberg Maritime CONTROS HydroFlash O2 optode", whos data sheet can be found here:
https://www.kongsberg.com/contentassets/4259c6dfed2d4ae3980fba37ec013ea8/hydroflash-accurate-fast-and-versatile-oxygen-optode.pdf

Is it possible to implement this instrument type?

gwemon, 2020-08-13:
Hello @hahn-johannes - yes indeed. Could you create a new issue ticket for it please? We are a little low on numbers at the moment due to staff being on leave so this will be looked at later in the month if that's okay with you?

hahn-johannes, 2020-08-13:
Hi @gwemon. Ok, I will do that and set a link to this issue to have a comparison to a quite similar instrument type (at least same company and both with some kind of an optical measurement principle).