nvs-vocabs/L22 #39: NTR:

rhudak2020, 2020-08-14:
Request that L22/TOOL0160 WETLabs C-Star transmissometer separate into two separate vocabs referring to the two different path lengths (25cm and 10cm). This is more of a "can you do this?".

roy-lowry, 2020-08-14:
A question that interests me as it's something I'm not sure I got right at the outset with L22. TOOL0003 is a single concept covering SeaTech transmissometers of multiple path lengths and multiple wavelengths. In the case of path lengths these are clearly different physical instruments. However, changing wavelength was a bit like changing a light bulb with LED cartridges.  If I had my time again I would probably have set up 4 concepts - the current TOOL0003 plus one for each path length.

Looking at the SeaBird documentation for the C-Star (https://www.seabird.com/transmissometers/c-star-transmissometer/family?productCategoryId=54627869913) there are clearly twelve different C-Star instruments with fixed properties, each with its own designation. For example, the C-Star FAS-000222 is a 530nm instrument with a 25cm path length and a depth rating of 600m. As these are so cleanly designated it would be possible to set up twelve new concepts in addition to the existing concept. This needs to be left unchanged to avoid compromising legacy data and provide a get-out clause for cases where the full instrument details are not known.

Other possibilities would be to set up two new entries (one for each path length), each covering six models or six new entries (one for each path length/wavelength combination) each covering two models. I'm not sure that I like either of these.

A note to @rhudak2020 is that these are just my thoughts and don't represent any commitment to do anything. There may also be other views in the team currently working on the vocab (I'm retired but active).

alex-tate, 2023-01-06:
Has there been any resolution on the C-Star L22 concepts? We have a 25cm path length, 650nm (red) wavelengh instrument rated to 6000m and are using the L22/TOOL0160 code but would prefer a more specific code if it were available. I agree with @roy-lowry that the path length is the most important factor, closely followed by the wavelength, with the depth rating more of a deployment characteristic.

rhudak2020, 2023-01-06:
@alex-tate I do not think a decision has been made on this request. I agree with @roy-lowry  and think new vocabs should be created for each path length and wavelength combination. In my opinion, depth is not as big of a key factor in the data collection/results. My reasoning is because pathlength and wavelength determine the type of data being collected. For example, you would want a 10cm pathlength in murky water and would want 650nm (red) for plankton.

louatbodc, 2023-01-06:
Apologies, it looks like this issue did not get automatically registered in our internal inventory (hence the current resolution). 

I completely agree since C-Stars are designated by their depth and wavelength (below). However, the pathlength is most critical to type of observation derived from the sensor. 

CST-236PR > Plastic unit, 600 meter rated, red 650nm
CST-1219DG > Deep unit, 6000 meter rated, green 532nm

Designator key:
P: Plastic unit rated to 600 meters
D: Deep unit rated to 6000 meters
R: Red, wavelength 650nm
G: Green, wavelength 532nm
B: Blue, wavelength 470nm

We will look into to completing this request as a priority

louatbodc, 2023-01-06:
I forgot to say that the wavelength is important towards the target particulate matter surveyed

roswri, 2023-02-06:
@rhudak2020, We have been discussing this, and we have decided that we are happy to create terms that are specific (down to path length, wavelength and depth rating) as long as there is a need for them. We wouldn't immediately create every possible combination path length, wavelength and depth rating because we can't be certain all the combinations are actually routinely manufactured and used. But if you have specific combinations that you would like us to create codes for we can do that.

@alex-tate, Would you like me to go ahead and create a new term for your WET Labs C-Star 25cm path-length, 650nm wavelength, 6000m depth rated transmissometer? Do you know if there's a specific model number

alex-tate, 2023-02-07:
@roswri, not sure it has a model number written on the sensor - it might have been on the original PO/invoice. However, I often use https://www.seabird.com/third-party-sensor-configuration to work out what things are according to Sea-Bird and this shows:

FAS-000224    Sea-Bird Scientific C-Star Transmissometer Deep (CST-25-DR), 25 cm path, 650 nm (red), 6000 m (replaces 24354).

Presumably CST-25-DR would be the specific model reference.

rhudak2020, 2023-02-15:
The only combinations we have at the moment are:
25cm pathlength, 650nm (red), 6000m 
10cm pathlength, 650nm (red), 6000m

JordanAthertonNOC, 2023-02-28:
Two new L22 terms created, these will be live on the NVS after ~07:30 GMT tomorrow:

WET Labs {Sea-Bird WETLabs} C-Star CST-25-DR transmissometer (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1902/)
WET Labs {Sea-Bird WETLabs} C-Star CST-10-DR transmissometer (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1903/)

closed 2023-02-28 by JordanAthertonNOC