nvs-vocabs/L22 #41: NTR: bulk term request batch 3

mcuthill, 2020-09-28:
Excel template and documentation attached.

[ONC_batch_3_submissions_v2.2.xlsx](https://github.com/nvs-vocabs/L22/files/5294571/ONC_batch_3_submissions_v2.2.xlsx)

Documentation:
[GeoSpectrum M8 hydrophone spec sheet.pdf](https://github.com/nvs-vocabs/L22/files/5294572/GeoSpectrum.M8.hydrophone.spec.sheet.pdf)
[GeoSpectrum M8E_M8E-51 hydrophone spec sheet.pdf](https://github.com/nvs-vocabs/L22/files/5294574/GeoSpectrum.M8E_M8E-51.hydrophone.spec.sheet.pdf)
[GeoSpectrum M8Q v2.pdf](https://github.com/nvs-vocabs/L22/files/5294575/GeoSpectrum.M8Q.v2.pdf)
[GeoSpectrum M36-100 v4.pdf](https://github.com/nvs-vocabs/L22/files/5294577/GeoSpectrum.M36-100.v4.pdf)
[Honeywell_HMR3300 Datasheet.pdf](https://github.com/nvs-vocabs/L22/files/5294578/Honeywell_HMR3300.Datasheet.pdf)
[HydroptyicUVP5Shallow USER MANUAL_20151217_SN103.pdf](https://github.com/nvs-vocabs/L22/files/5294580/HydroptyicUVP5Shallow.USER.MANUAL_20151217_SN103.pdf)
[Ifremer_Chemini_datasheet.pdf](https://github.com/nvs-vocabs/L22/files/5294581/Ifremer_Chemini_datasheet.pdf)
[Keithley Instruments_M1000_2000.pdf](https://github.com/nvs-vocabs/L22/files/5294582/Keithley.Instruments_M1000_2000.pdf)
[Max Planck Institute_SMP Sediment_Microprofiler.pdf](https://github.com/nvs-vocabs/L22/files/5294584/Max.Planck.Institute_SMP.Sediment_Microprofiler.pdf)
[Moog_Crossbow_NAV440_Datasheet_revB.pdf](https://github.com/nvs-vocabs/L22/files/5294586/Moog_Crossbow_NAV440_Datasheet_revB.pdf)
[Naxys Hydrophone 02345 DataSheet 2012-02-29.pdf](https://github.com/nvs-vocabs/L22/files/5294588/Naxys.Hydrophone.02345.DataSheet.2012-02-29.pdf)
[NRCAN BPR Specifications.pdf](https://github.com/nvs-vocabs/L22/files/5294589/NRCAN.BPR.Specifications.pdf)
[PNI_TCM2.6_Datasheet_05-14-2009.pdf](https://github.com/nvs-vocabs/L22/files/5294590/PNI_TCM2.6_Datasheet_05-14-2009.pdf)
[Reftek 130-01 seismic recorder_manual.pdf](https://github.com/nvs-vocabs/L22/files/5294591/Reftek.130-01.seismic.recorder_manual.pdf)
[UMNGasTightSampler_Jan2017_ONC_Instrument_Specs_Report.pdf](https://github.com/nvs-vocabs/L22/files/5294592/UMNGasTightSampler_Jan2017_ONC_Instrument_Specs_Report.pdf)
[UW_BARS_User manual.pdf](https://github.com/nvs-vocabs/L22/files/5294593/UW_BARS_User.manual.pdf)
[WA Song Meter SM4M datasheet.pdf](https://github.com/nvs-vocabs/L22/files/5294594/WA.Song.Meter.SM4M.datasheet.pdf)

danibodc, 2021-03-23:
The following terms from this batch have now been loaded to L22. Please note they will become live on the NVS tomorrow morning after the overnight database refresh:

L22	TOOL1661	GeoSpectrum M36-100 hydrophone
L22	TOOL1660	GeoSpectrum M8Q hydrophone
L22	TOOL1659	GeoSpectrum M8E-51 hydrophone
L22	TOOL1658	GeoSpectrum M8 hydrophone
L22	TOOL1657	Keithley Instruments M1000/M2000 data logger series
L22	TOOL1654	University of Minnesota [custom] Gas-Tight serial sampler - Wu et al. (2015)
L22	TOOL1653	RBR Cervata (submersible) data logger
L22	TOOL1652	Natural Resources Canada, Geological Survey of Canada [custom] Bottom Pressure Recorder
L22	TOOL1651	University of Washington [custom] Benthic and Resistivity Sensors (BARS)
L22	TOOL1650	Hydroptic Underwater Vision Profiler 5 Shallow {UVP5-Shallow} imaging sensor
L22	TOOL1649	4-Deep Inwater Imaging HoloSea S7 submersible holographic microscope
L22	TOOL1597	Sound Metrics Aris Explorer 3000 imaging sonar
L22	TOOL1596	Sonardyne Fetch 8306 sensor logging node
L22	TOOL1595	Silicon Audio Low-Noise optical seismometer
L22	TOOL1594	Ref Tek 130-01 broadband seismic recorder
L22	TOOL1593	Naxys Ehyd 02345 ethernet hydrophone
L22	TOOL1592	LEMI LLC LEMI-039 3C 3-component analog magnetometer
L22	TOOL1591	Kistler K-Beam 8395A capacitive MEMS triaxial accelerometer series
L22	TOOL1590	JFE Advantech Rinko II {ARO-CAR/CAD} oxygen sensor
L22	TOOL1589	Guralp Systems CMG-1T ocean bottom seismometer

We are continuing to work on the following remaining terms from this batch. Thank you for your patience:

Max Planck Institute SMP [custom] sediment microprofiler
Honeywell Aerospace HMR 3300 (3-axis tilt-compensated) electronic compass
PNI Sensor Corp TCM 2.6 (3-axis tilt-compensated) electronic compass
MOOG Crossbow NAV440 inertial system series
Ifremer Chemical Miniaturised Analyser Fe {CHEMINI} (deep sea version) iron analyser
Geospace Technologies GS 11D 3C [modified] geophone
Nanometrics Trillium 120 Posthole seismometer
Wildlife Acoustics Song Meter SM4M marine recorder series
Technicap PPS 3/3-24S sediment trap
JASCO AMAR Observer G3 autonomous multichannel acoustic recorder

louatbodc, 2021-03-23:
The following terms have been published:

Honeywell Aerospace HMR 3300 (3-axis tilt-compensated) electronic compass (TOOL1662)
PNI Sensor Coorporation TCM 2.6 (3-axis tilt-compensated) electronic compass (TOOL1663)
MOOG Crossbow NAV440 heading, attitude and positioning sensor (TOOL1664)
Ifremer Chemical Miniaturised Analyser deep sea version Fe {CHEMINI} iron analyser (TOOL1665)
Nanometrics Trillium 120 Posthole 120 PH seismometer (TOOL1666)
Wildlife Acoustics Song Meter SM4M underwater sound recorder series (TOOL1667)
Technicap PPS 3-3 24S sediment trap series (TOOL1668)

However, we require input from you on the following terms:

Max Planck Institute SMP [custom] sediment microprofiler
This looks like a unit which can house different types of sensors and could potentially be considered a platform. The sensors fitted are not constrained. Thus we would recommend using terms to describe the individual sensors.

Geospace Technologies GS 11D 3C [modified] geophone
I unfortunately can't find any information relating to this instrument. Please could you provide a description? In particular, how this instrument has been modified from the manufacturer design. Also, please describe what makes a 3C different to just the 11D series.

JASCO AMAR Observer G3 autonomous multichannel acoustic recorder
It is not too clear if this instrument is essentially a data logger that can house a variety of sensors from manufacturers. If the sensor houses a variety of sensors from different manufacturers it becomes difficult to constrain in a controlled term and we might recommend using concepts for the individual sensors. Do you have other information about this sensor? For example. what sensors it was using and how the Observer differs from other models in this series?

mcuthill, 2021-06-02:
Re the JASCO AMAR Observer G3 autonomous multichannel acoustic recorder:
It appears the 'Observer' term was added erroneously by some internal naming convention - these devices are the standard JASCO AMAR G3 devices. 
Our instrument point person for these devices said "For sensors of the AMAR G3 device it just has the standard health sensors (battery voltage/current, temperature) and pitch/roll/yaw. It's mainly used with JASCO hydrophone arrays to support data acquisition." 
This supports the idea that it's basically a data logger, specifically used for acoustic instruments. I've attached the user guide with specs that we have on record for the device: 
[AMAR G3 User Guide.pdf](https://github.com/nvs-vocabs/L22/files/6587016/AMAR.G3.User.Guide.pdf)
I hope this clarifies things!

mcuthill, 2022-01-14:
I just noticed there's a typo in PNI Sensor **Coorporation** TCM 2.6 (3-axis tilt-compensated) electronic compass (TOOL1663). Portmanteau for a cooperative corporation perhaps?

danibodc, 2022-01-20:
Thank you for pointing this out @mcuthill, this has now been updated:

PNI Sensor Corporation TCM 2.6 (3-axis tilt-compensated) electronic compass

The change will be live on the NVS from 0730 UTC tomorrow.

JordanAthertonNOC, 2023-06-26:
Hi @mcuthill 

Apologies for the long delay in resolving this ticket - I believe the only outstanding term from this request is the JASCO AMAR G3, I have now created this, and it should be live on the NVS from ~07:30 UTC tomorrow:

JASCO AMAR G3 acoustic recorder (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1966/)

Feel free to reopen the ticket if we've missed anything or if further information comes to light regarding the Geospace Technologies GS 11D 3C.

closed 2023-06-26 by JordanAthertonNOC