nvs-vocabs/L22 #52: NTR: PQS1 PAR Quantum Sensor

botheredbybees, 2020-11-17:
## Device make
Kipp and Zonen

## Device range
n/a

## Device model
PQS1

## Model version
[Name of the sub-model (e.g. 'D', 'v3.2'). If there is no official version/sub-model then input 'n/a'.]

## Model alternative name (if applicable)
PQS1  

## Qualification (if applicable)
PAR Quantum Sensor

## Instrument type
Photosynthetically Active Radiation sensor

## Does the tool describe a series of instruments? 
no

## Short name
Kipp and Zonen PQS1 PAR

## Documentation
https://www.kippzonen.com/Product/184/PQS1-PAR-Quantum-Sensor#.X7RBbWhLhZc

## Mapping to external terminologies (optional)
Broader | http://vocab.nerc.ac.uk/collection/L05/current/122/

## ORCID (optional)
https://orcid.org/0000-0001-6840-6785

roswri, 2021-01-20:
Hi Peter,

There was a term for this already in L22 but there was a mistake in the name so I have corrected this now: 
Kipp and Zonen PQS1 PAR Quantum Sensor (http://vocab.nerc.ac.uk/collection/L22/current/TOOL0530/)
Please note, the web-page may take a little while to update with the new preferred and alternate labels.

Best wishes,
Roseanna

closed 2021-01-20 by roswri

gwemon, 2021-01-20:
@botheredbybees Just to say that the update has been queued for publication but is not yet live. We currently have only one publication round per day and the next one will be shortly after 0700UT (AM) tomorrow.