nvs-vocabs/L22 #55: NTR: SBE-49 FastCAT CTD

rhudak2020, 2020-11-23:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
SeaBird

## Device range
n/a

## Device model
SBE-49

## Model version
n/a

## Model alternative name (if applicable)
FastCAT CTD

## Qualification (if applicable)
[Qualification of the instrument type i.e. a word or words that further describe the instrument type (e.g. 'portable','automated','pressure-compensated','large' etc.). Sometimes these are used in manufacturer names.]

## Instrument type
CTD

## Does the tool describe a series of instruments? 
[Does the term describe a manufacturing 'series' of instruments, a 'system' of instruments, an 'individual' instrument or other?]

## Reference (if applicable)
[Citation for the instrument (e.g. Hall and Bumpus 1937, 1944). Use if the instrument is manufactuered and built according to a design in a paper.]

## Short name
SBE-49

## Documentation
https://www.seabird.com/moving-platform/sbe-49-fastcat-ctd-sensor/family?productCategoryId=54627473793

## Mapping to external terminologies (optional)
broader:http://vocab.nerc.ac.uk/collection/L05/current/130/

## ORCID (optional)
0000-0002-0017-8330

danibodc, 2021-01-04:
Hello, many thanks for your request. This instrument already appears in L22 at the following URL:

http://vocab.nerc.ac.uk/collection/L22/current/TOOL0827/

There appears to be a typo in the preferred label however; it currently states 'FasCAT' as opposed to FastCAT. I will correct this now and let you know once it has been re-published.

danibodc, 2021-01-05:
The update above is now live: http://vocab.nerc.ac.uk/collection/L22/current/TOOL0827/

Many thanks!

closed 2021-01-05 by danibodc