nvs-vocabs/L22 #57: Bulk New Term Request

rhudak2020, 2020-12-15:
Attaching template for bulk submission of terms. 

[20200813_L22_submission.xlsx](https://github.com/nvs-vocabs/L22/files/5696853/20200813_L22_submission.xlsx)

roswri, 2021-09-06:
Hi @rhudak2020 ,

We are nearly done creating terms. I just have a quick question about the 'Dynamic Gravity Systems AT1 gravimeter', I assume this is the marine version, could you confirm?

Many thanks,
Roseanna

rhudak2020, 2021-09-07:
Hi @roswri,

Yes the Gravimeter is a marine version/installation. Thank you.

Rebecca

louatbodc, 2022-11-01:
Thank you for your request. I'm pleased to say we have now completed the publication of terms. The terms can be viewed here:

https://github.com/nvs-vocabs/L22/issues/170

Many thanks
Lou

closed 2022-11-01 by louatbodc