nvs-vocabs/L22 #68: NTR: Vaisala PTB330

tkeena, 2021-03-08:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Vaisala

## Device range
Barometer PTB330

## Device model
PTB330

## Model version

## Model alternative name (if applicable)

## Modification (if applicable)
[Describes any modifications to the instrument design, such as a manufactured device that has been modified 'in-house' or a plakton net that has been modified since its first description.]

## Qualification (if applicable)
[Qualification of the instrument type i.e. a word or words that further describe the instrument type (e.g. 'portable','automated','pressure-compensated','large' etc.). Sometimes these are used in manufacturer names.]

## Instrument type
Barometer 

## Does the tool describe a series of instruments? 
[Does the term describe a manufacturing 'series' of instruments, a 'system' of instruments, an 'individual' instrument or other?]

## Reference (if applicable)
[Citation for the instrument (e.g. Hall and Bumpus 1937, 1944). Use if the instrument is manufactuered and built according to a design in a paper.]

## Short name
[Shortened name of the instrument. This is usually a shortened version of name limited to make, model and version. It can also be an alternative name.]

## Documentation
https://www.vaisala.com/en/products/instruments-sensors-and-other-measurement-devices/weather-stations-and-sensors/ptb330

## Mapping to external terminologies (optional)
[Is your submission related to any existing terms in external resources: please enter one or a list of URIs]

## ORCID (optional)
[Enter your ORCID if desired]

rhudak2020, 2021-03-08:
Hi @tkeena, there is a L22 term for the Vaisala PTB330 http://vocab.nerc.ac.uk/collection/L22/current/TOOL1015/ . Are you proposing a more specific model?

tkeena, 2021-03-09:
Many thanks for your quick response. That term is perfect. My apologies - I initially could only find the PTB110. This request can be closed.

gwemon, 2021-03-09:
Thank you @rhudak2020 and @tkeena. I am closing this ticket.

closed 2021-03-09 by gwemon