nvs-vocabs/L22 #69: NTR: Vaisala PTU307

rhudak2020, 2021-04-13:
## Device make
Vaisala

## Device range
n/a

## Device model
PTU

## Model version
307

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
Combined Pressure, Humidity and Temperature Transmitter

## Instrument type
met station (air temp, barometer, relative humidity)

## Does the tool describe a series of instruments? 
PTU300 series

## Reference (if applicable)
n/a

## Short name
PTU307

## Documentation
https://www.vaisala.com/sites/default/files/documents/PTU300-Datasheet-B210954EN.pdf

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
0000-0002-0017-8330

danibodc, 2021-04-20:
Hi,

We have the following L22 term that covers the PTU300 series of instruments. The PTU307 is included in the description. Hopefully this is suitable for your purpose but if not let us know. I could also update the preferred label to include 'series' if this would help:

http://vocab.nerc.ac.uk/collection/L22/current/TOOL0741/

Preferred Label: 
Vaisala PTU300 Pressure, Humidity and Temperature Transmitter

Definition: 
The Vaisala PTU300 is a series of air pressure, humidity and temperature transmitters comprising a base station and sensors that may be attached directly or on a remote probe suited to the application. The probes are designed for installation in various environments: PTU301 for laboratories, PTU303 for general use and PTU307 for outdoor and meteorological applications (warmed probe). All incorporate the Vaisala BAROCAP pressure sensor, a thermistor and the Vaisala HUMICAP humidity sensor giving pressure accurate to 0.1-0.3 hPa (depends upon sensor class), temperature accurate to 0.2C and humidity accurate to 1%. The base station includes a graphical display giving continuous digital readout and keypad for operation control. Depending upon the options selected data may be logged internally (up 4 years) or transmitted via wired LAN, wireless LAN, USB, RS232 or RS485. More information is available at http://www.vaisala.com/Vaisala%20Documents/Brochures%20and%20Datasheets/CEN-TIA-G-PTU300-Combined-Brochure-B210954EN-E-LOW-v6.pdf.
--

The Vaisala PTU300 is a series of air pressure, humidity and temperature transmitters comprising a base station and sensors that may be attached directly or on a remote probe suited to the application. The probes are designed for installation in various environments: PTU301 for laboratories, PTU303 for general use and PTU307 for outdoor and meteorological applications (warmed probe). All incorporate the Vaisala BAROCAP pressure sensor, a thermistor and the Vaisala HUMICAP humidity sensor giving pressure accurate to 0.1-0.3 hPa (depends upon sensor class), temperature accurate to 0.2C and humidity accurate to 1%. The base station includes a graphical display giving continuous digital readout and keypad for operation control. Depending upon the options selected data may be logged internally (up 4 years) or transmitted via wired LAN, wireless LAN, USB, RS232 or RS485. More information is available at http://www.vaisala.com/Vaisala%20Documents/Brochures%20and%20Datasheets/CEN-TIA-G-PTU300-Combined-Brochure-B210954EN-E-LOW-v6.pdf.
--

rhudak2020, 2021-04-27:
Hi, 

I think that a series should suffice for this term, especially since in this case the differences between instruments are so small. I think it might be worth discussing for terms/series that are different in terms of data output, etc. Worth thinking about anyway, thanks for the reply.

closed 2021-04-27 by danibodc