nvs-vocabs/L22 #76: NTR: New L22 terms published as part of BCO-DMO request via email [BODCNVS-595]

louatbodc, 2021-04-28:
The following terms were submitted by BCO-DMO via email request (reference [BODCNVS-595]). The terms were investigated and published as follows:

**Tools published:**

TOOL1605 - Rotronic Hygromer MP101 A temperature and humidity probe
TOOL1606 - Navico Simrad MX421 {MX Marine MX421} GPS series
TOOL1607 - Sperry Marine NAVIGAT 2100 gyrocompass
TOOL1608 - Sperry Marine NAVIGAT 3000 gyrocompass
TOOL1609 - Orolia Spectracom Netclock 9383 GPS clock synchroniser
TOOL1610 - Krohne Optiflux 5000 flow sensor
TOOL1611 - Vaisala PTB101 B barometer
TOOL1612 - AML Oceaographic P-Xchange pressure sensor series
TOOL1613 - Sunburst Sensors Submersible Autonomous Moored Instrument pH {SAMI-pH} seawater and freshwater pH sensor series
TOOL1614 - Sea-Bird SBE 3S temperature sensor
TOOL1615 - Kongsberg Seatex Seapath 380 5+ global positioning and attitude sensor
TOOL1616 - Marine Magnetics Seaspy 2 magnetometer
TOOL1617 - Kongsberg Seatex MRU 5 platform attitude sensor
TOOL1618 - Orolia Spectracom SecureSync clock synchroniser
TOOL1619 - Simrad GC80 {gyro} gyrocompass
TOOL1620 - Kongsberg Simrad HS50 GPS gyrocompass
TOOL1621 - Furuno GPS Navigator GP-31 GPS
TOOL1622 - Turner Designs Cyclops-7 fluorometer
TOOL1547 - Furuno GP-330B GPS/WAAS receiver antenna
TOOL1548 - Biospherical GUV-2511 Ground Based UV/PAR radiometer
TOOL1549 - Vaisala HMP 110 humidity and temperature probe
TOOL1550 - Vaisala HMP 45 humidity and temperature probe
TOOL1551 - HOBI Labs HydroDAS data logger
TOOL1552 - Japan Radio Co (JRC) JLN-550 doppler sonar
TOOL1553 - Furuno GP-320B GPS/WAAS receiver antenna
TOOL1554 - Furuno GP-33 GPS navigator
TOOL1555 - Kongsberg K-Pos DP-11 dynamic positioning system
TOOL1556 - Wärtsilä ELAC LAZ 5100 single-beam echosounder
TOOL1557 - Columbia Weather Magellan MX Weather Station series
TOOL1558 - Teledyne TSS Meridian Standard gyrocompass
TOOL1559 - Gill MetPak Pro Weather Station
TOOL1624 - Thales DG16 Differential Global Positioning System receiver
TOOL1625 - Forsberg Starlink DNAV-212 Differential Global Positioning System receiver
TOOL1626 - Kongsberg Seatex Differential Positioning Sensor 112
TOOL1627 - Trimble DSM 232 Global Positioning System receiver
TOOL1628 - Raytheon DSN-450 Doppler speed log
TOOL1629 - WETLabs ECO-FLNTU(S) combined fluorometer and turbidity sensor
TOOL1630 - Skipper Electronics EML224 electromagnetic speed log
TOOL1631 - Coda Octopus F185 gyrocompass
TOOL1632 - Furuno FCV 585 video echosounder fish finder
TOOL1633 - Furuno Navigator GP170 Global Positioning System receiver
TOOL1634 - Furuno SC-30 GPS receiver and gyrocompass
TOOL1635 - Omega FP-2541 flow meter
TOOL1636 - ESE ES-185 A Global Positioning System master clock
TOOL1637 - ESE ES-185 E Global Positioning System master clock
TOOL1516 - RM Young 41342 temperature probe
TOOL1517 - RM Young 41372 hygrometer
TOOL1518 - RM Young 85000 ultrasonic anemometer
TOOL1519 - RM Young 85106 ultrasonic anemometer
TOOL1520 - RM Young 86106 ultrasonic anemometer
TOOL1521 - Trimble ABX-Two gyrocompass
TOOL1522 - AML Oceaographic C-Xchange salinity sensor
TOOL1524 - Ashtech ADU800 gyrocompass
TOOL1525 - FloCat C-ES45 B002 flow meters
TOOL1526 - Heitronics CT15 thermometer

**The following were not published as they are control and display units or serial interfaces:**
Navico Simrad MX51X {MX Marine MX51X} GPS control and display unit
Simrad MX612 {MX Marine MX612} GPS control and display unit
Young 32400 - described as a magnetic compass but is actually a serial interface for the 32500 model electronic compass which already has an L22 term (TOOL1327), so not sure if this actually needs to be its own L22, since a Serial Interface isn't a sensor or sample collecting device I'm not sure L22 is appropriate

**The following were not published as they are XBT launchers:**
Sippican MK12 XBT launcher
Sippican MK12 XBT launcher

**The following already existed:**
McLane Research Laboratories Imaging FlowCytobot {IFCB} Imaging flow cytometer = TOOL1588
Sea-Bird Laboratories SBE-43 dissolved oxygen sensor = TOOL0036
AML Oceanographic Smart-X multi-parameter probe = TOOL1331 (note we have updated the description to make this clear)
Valeport miniSVS sound velocity sensor = TOOL1495
Young 5106 = TOOL0195
Young 41382 = TOOL1000
Young 61302 = TOOL1001

rhudak2020, 2021-04-29:
Does NERC not publish XBT launchers? Or is it under a different category?

JordanAthertonNOC, 2023-02-08:
Hi @rhudak2020, apologies for the delay in addressing this, it dropped off our internal tracking procedure and has only come to light again during a bit of housekeeping. 

We do not publish L22 terms for launchers, but we do for systems which involve a data collection element. Is this here request for just a launcher element or does it form a system (e.g. Sippican MK12 XBT system)? If the latter do you have further information about this device?

Thanks in advance.

JordanAthertonNOC, 2023-03-08:
We are closing this ticket, feel free to re-open if additional information comes to light.

closed 2023-03-08 by JordanAthertonNOC