nvs-vocabs/L22 #78: duplicate instruments

rhudak2020, 2021-05-05:
Hi,

I noticed there seems to be a duplication. There could be an explanation, but I wanted to start a conversation for clarification. There were two submissions for Vaisala HMP155, each created a term:

http://vocab.nerc.ac.uk/collection/L22/current/TOOL1673/ - "Vaisala HMP 155 hygrometer series"
http://vocab.nerc.ac.uk/collection/L22/current/TOOL1687/ - "Vaisala HUMICAP HMP155 Humidity and Temperature Probe"

The second term specifies the Vaisala HUMICAP 180R sensor which is included in the 'series' term. In my opinion (and I'm looking for any other thoughts), if the second term is going to specify the 180R sensor then that should be in the name/label and not only in the description.

Cheers,

Becca

danibodc, 2021-09-07:
Hi @rhudak2020 
Many thanks for this, yes we agree with your statement. The 'series' term covers all the various HUMICAP 180 sensors that can be used as part of the HMP155. 

As http://vocab.nerc.ac.uk/collection/L22/current/TOOL1687/ is a term describing the specific use of the HUMICAP 180R sensor, I have updated the preferred label and alternative label for this term to include the 180R.

This should be live on the NVS from 0730 UTC tomorrow morning.

Thanks again!

rhudak2020, 2021-12-09:
@danibodc I recently received some more info about the HUMICAP variations, there are 3. 

**Vaisala HUMICAP® 180C** is a composite general purpose humidity sensor with integral Pt100 sensor. It has good humidity durability and short response time. The sensor contains both humidity and temperature sensors. This sensor is used for chemical purge and heated probe options.
**Vaisala HUMICAP® 180R** is a robust general purpose humidity sensor. It has high humidity durability, good long-term stability and good tolerance against chemical exposure.
**Vaisala HUMICAP® 180RC** is a robust composite general purpose humidity sensor. It has high humidity durability, good long-term stability and good tolerance against chemical exposure. This sensor includes an integral Pt100 temperature sensor and is used for both the chemical purge and heated probe units.

It all boils down to if it has a temperature sensor and it’s durability and stability, classified as composite (warmed probe and composite sensor), robust (no warmed probe but robust sensor), or robust composite (warmed probe and robust sensor).

I'd like to suggest a term for each of the HUMICAP variations.

rhudak2020, 2021-12-13:
We received some more new information from Vaisala regarding these different models:

These are different generations of HUMICAPs.
180->180R->R2

We [Vaisala] are doing continuous improvement. 180R has better long term stability than 180. R2 has better corrosion resistance than 180R. C standards for composite. For heated HMP155, always use composite sensor as the PT100 is glued to HUMICAP and acts as a heater. Temperature measurement comes from additional T probe.

Most reliable sensor with best long term stability: R2
Fast humidity response: INTERCAP
Heated probe: composite

 
[image]

JordanAthertonNOC, 2023-05-16:
Two new L22's have been created for the 180C and 180RC respectively to complement the existing 180R:

Vaisala HUMICAP 180C HMP155 humidity and temperature probe (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1916/)
Vaisala HUMICAP 180RC HMP155 humidity and temperature probe (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1917/)

closed 2023-05-16 by JordanAthertonNOC