nvs-vocabs/L22 #87: Is it correct to have a "sameAs" relationship withing hierarchically related vocabualries like L05 and L22?

gwemon, 2021-07-01:
This is creating some anomalies in the display of the [SDN Device scheme under the new NVS VocPrez UI](http://vocab.nerc.ac.uk/scheme/SDNDEV/current/) and I think that it is worth raising a ticket for this because it raises the question of declaring an equivalence between concepts based solely on the preferred label.

We have 3 such cases in the SDNDEV schemes where the relationship "sameAs" has been used to link:
1- an unknown instrument http://vocab.nerc.ac.uk/collection/L22/current/TOOLZZZ/ to the unknown instrument category http://vocab.nerc.ac.uk/collection/L05/current/999/
2- GPS generic instrument L22 concept http://vocab.nerc.ac.uk/collection/L22/current/TOOL0078/ to the category http://vocab.nerc.ac.uk/collection/L05/current/POS03/
3- Hydrophone generic instrument L22 concept http://vocab.nerc.ac.uk/collection/L22/current/TOOL0354/ to the category http://vocab.nerc.ac.uk/collection/L05/current/369/

My argument is that if one represents the model of an instrument instance and the other one is a category of instruments then the latter concept should always be broader than the former even if the former is a generic term.

Can we go ahead and change the sameAs to a broader/narrower relationship? As we do so it might be necessary to review the labelling of the L22 term.

roy-lowry, 2021-07-01:
Hi Gwen,

Your argument makes perfect sense. An instance and a instance classification cannot have a sameAs mapping so definitely an error (probably mine :-))

rhudak2020, 2021-07-16:
Hi Gwen,

Not sure if you are still taking comments on this, but I would also agree with your logic. The narrow/broader makes sense for a category -> model relationship.

gwemon, 2021-07-19:
Thank you @rhudak2020 and @roy-lowry - I have made the changes and they will be live tomorrow.

closed 2023-02-08 by gwemon