nvs-vocabs/L22 #92: NTR: Seatex MGC R3

rhudak2020, 2021-10-05:
## Device make
Kongsberg

## Device range
n/a

## Device model
MGC

## Model version
R3

## Model alternative name (if applicable)
MGC R3

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
gyrocompass / ins

## Does the tool describe a series of instruments? 
"family" of instruments

## Reference (if applicable)
n/a

## Short name
MGC R3

## Documentation
https://www.kongsberg.com/maritime/products/bridge-systems-and-control-centres/navigation-system/gyro-compass-systems/gyro-compass-and-ins-mgc-r3/

## Mapping to external terminologies (optional)
[Is your submission related to any existing terms in external resources: please enter one or a list of URIs]

## ORCID (optional)
0000-0002-0017-8330

danibodc, 2023-02-27:
Hi @rhudak2020 

This term has now been loaded to L22 and will be live on the NVS from tomorrow morning:

TOOL1900	Kongsberg Seatex MGC R3 gyrocompass inertial navigation system

danibodc, 2023-02-28:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1900/

closed 2023-02-28 by danibodc