nvs-vocabs/L22 #102: NTR: avvistApp

meliezer, 2021-11-30:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Divulgando Srl

## Device range
n/a

## Device model
n/a

## Model version
n/a

## Instrument type
App

## Does the tool describe a series of instruments? 
individual, but two versions: Android and iOS.

## Short name
avvistApp

## Documentation
How citizens can contribute to the monitoring of marine animals with smartphones. This app is freely available on Google and Apple stores. https://doi.org/10.13120/h127-9v54

SLBlakeman, 2023-04-11:
Good afternoon @meliezer,

Thank you for requesting a new L22 term, unfortunately this vocabulary is limited to instruments and we cannot create terms for software. 

We could create a term for a "mobile phone camera" which would be the instrument citizen scientists use to take photographic evidence of a sighting/location of a sighting. Please let me know if this is of interest to you.

Kind regards,
Samantha Blakeman

SLBlakeman, 2023-04-21:
A new term has been created for Mobile device camera. It will be available on the NVS 22/04/2023 at 08:00.

TOOL1910 - Unspecified mobile device camera

closed 2023-04-21 by SLBlakeman