nvs-vocabs/L22 #125: NTR: NORBITNORBIT WBMS Full Ocean Depth 12003-AFCDB4 400kHz

danropeNOC, 2022-05-24:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
NORBIT

[This is the name of the manufacturer or institution which made the instrument. If there is a manufacturer but you have not been able to confirm this information or there is no official manufacturer, input 'Inapplicable'. If you are creating a generic instrument (e.g. a generic temperature sensor, then input 'Unspecified'.]

## Device range
WBMS

[The manufacturer's instrument model range (e.g. WindObserver). If there is no official model range then input 'n/a'.]

## Device model
NORBIT WBMS Full Ocean Depth 12003-AFCDB4 400kHz

[Name of the instrument model. If there is no official model name, input 'n/a'.]

## Model version
N/A

[Name of the sub-model (e.g. 'D', 'v3.2'). If there is no official version/sub-model then input 'n/a'.]

## Model alternative name (if applicable)
N/A
[Alternative name for the model. E.g. Glider Payload CTD alternative name is GPCTD]

## Modification (if applicable)
N/A
[Describes any modifications to the instrument design, such as a manufactured device that has been modified 'in-house' or a plakton net that has been modified since its first description.]

## Qualification (if applicable)
Multi Beam Sonar System

[Qualification of the instrument type i.e. a word or words that further describe the instrument type (e.g. 'portable','automated','pressure-compensated','large' etc.). Sometimes these are used in manufacturer names.]

## Instrument type
Multi Beam Sonar

[Please indicate what type of instrument it is (e.g. global positioning system, fluorometer, etc.)]

## Does the tool describe a series of instruments? 
Individual Instrument
[Does the term describe a manufacturing 'series' of instruments, a 'system' of instruments, an 'individual' instrument or other?]

## Reference (if applicable)
N/A
[Citation for the instrument (e.g. Hall and Bumpus 1937, 1944). Use if the instrument is manufactuered and built according to a design in a paper.]

## Short name
NORBIT WBMS
[Shortened name of the instrument. This is usually a shortened version of name limited to make, model and version. It can also be an alternative name.]

## Documentation
[PS-120005-21_WBMS-bathy_Pn_12003-AACDB4_A4.pdf](https://github.com/nvs-vocabs/L22/files/8760325/PS-120005-21_WBMS-bathy_Pn_12003-AACDB4_A4.pdf)

[Please enter a link to the documentation describing the instrument. This could be a journal reference, manufacturer data sheet, web site, manual etc.]

## Mapping to external terminologies (optional)
[Is your submission related to any existing terms in external resources: please enter one or a list of URIs]

## ORCID (optional)
[Enter your ORCID if desired]

JordanAthertonNOC, 2022-06-08:
Hi @danropeNOC 

Could you clarify for me the model ID for this instrument? The request is for a 12003-**AF**CDB4, but the pdf provided has 12003-**AA**CDB4 in the filename. To confuse things further, the [NORBIT instrument configurator](https://norbit.com/subsea/configurator/) for a full ocean depth (up to 6000 m) option suggests a model ID of 12003-**RF**CDB4. If this represents a custom build, are you able to detail which configuration/features were selected? 

Thanks in advance.

danropeNOC, 2022-06-17:
> Caution: This email has originated from outside of the organisation. Do not click links or open attachments unless you have verified the sender and content is safe. Thank you.

Hi Jordan

12003-AFCDB4 was the part number supplied to us in the original quote from Norbit, the parts are marked # 24030-1-FCDB, although manufacturers often have different numbers for the full system ( inc cables accessories etc ) vs the sensor hardware itself.

Kind Regards
Dan

This email and any attachments are intended solely for the use of the named recipients. If you are not the intended recipient you must not use, disclose, copy or distribute this email or any of its attachments and should notify the sender immediately and delete this email from your system.
The National Oceanography Centre (NOC) has taken every reasonable precaution to minimise risk of this email or any attachments containing viruses or malware but the recipient should carry out its own virus and malware checks before opening the attachments. NOC does not accept any liability for any losses or damages which the recipient may sustain due to presence of any viruses.
Opinions, conclusions or other information in this message and attachments that are not related directly to NOC business are solely those of the author and do not represent the views of NOC.

roswri, 2022-09-27:
New term created:
NORBIT WBMS 12003-AFCDB4 400kHz (full ocean depth) multibeam echosounder (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1850/)

Please note the new term has been queued for publication but is not yet live. We currently have only one publication round per day and the next one will be shortly after 0700UT (AM) tomorrow.

closed 2022-10-04 by roswri