nvs-vocabs/L22 #130: NTR: New terms loaded to L22 (BODCNVS-1403)

danibodc, 2022-05-31:
The following term has been loaded to L22 and will be live on the NVS from 07:00 UTC tomorrow morning:

TOOL1806	Systea MicroMac 1000 portable analyser
TOOL1807	Turner Designs 10AU fluorometer
TOOL1808	Endress and Hauser TurbiMax W CUS 41 in-situ particle sizer

rhudak2020, 2022-06-15:
@danibodc can you determine the difference between TOOL1807 and TOOL0393 ..they both seem to be a Turner 10AU.

danibodc, 2022-07-14:
Hi @rhudak2020 

Thanks for this, we have determined that https://vocab.nerc.ac.uk/collection/L22/current/TOOL1807/ was created as a duplicate in error, so we have deprecated this term. We have updated https://vocab.nerc.ac.uk/collection/L22/current/TOOL0393 with the description from TOOL1807 as it contained some more detailed information. Please use https://vocab.nerc.ac.uk/collection/L22/current/TOOL0393.

The updates should be live from tomorrow morning.

Many thanks and apologies for the error!