nvs-vocabs/L22 #132: NTR: Markey COM-7 winch

rhudak2020, 2022-06-28:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Markey

## Device range
n/a

## Device model
COM-7

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
marine research

## Instrument type
winch

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
n/a

## Short name
COM-7

## Documentation
https://pdf.nauticexpo.com/pdf/markey-machinerympany/com-7e-v18414/28204-38001.html

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
0000-0002-0017-8330

SLBlakeman, 2022-08-12:
Hello @rhudak2020,

Thank you for submitting a request for a new term. Unfortunately this term does not meet the conditions of our L22 vocabulary, which is designed to contain instruments and sensors that are used to produce data, nor our B76 vocabulary for platforms.

Please do let me know if you have any questions on this matter.

Kind regards,
Samantha Blakeman

rhudak2020, 2022-08-12:
Hello @SLBlakeman,

Apologies for my mistake. I had thought I had seen some other Markey models in the L22 vocabulary, although now I do not see any and I'm not sure where/what I was looking at. Thank you for correcting me.

SLBlakeman, 2022-08-12:
Hello @rhudak2020,

No apology necessary. I've had a look and we do have the manufacturer in our database, however I can't find any associated instruments. Sorry I can't be of more help. 
I shall close this ticket now as no further action is required.

closed 2022-08-12 by SLBlakeman