nvs-vocabs/L22 #138: NTR: IFREMER mesopelagos net

JoBeja, 2022-07-13:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
IFREMER custom made net

## Device range
n/a

## Device model
n/a

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
2.	12.6m long and with a mouth opening of 8.4m in total
3.	Varying mesh size from the top to bottom of the net, at specific lengths (see diagram)
4.	Rectangular fixed frame at the top measuring 5.6m*1.4m(?)
5.	Opening mouth area=50m^2
6.	Can be towed oblique-horizontal or oblique tow at 2knots

## Qualification (if applicable)
n/a

## Instrument type
net

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
See Chalut_Grand_Mesopelagos.pdf which will be sent via email to the vocab group email

## Short name
Mesopelagos net

## Documentation
See Chalut_Grand_Mesopelagos.pdf which will be sent via email to the vocab group email

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
https://orcid.org/0000-0002-5196-8447

danibodc, 2022-09-30:
HI @JoBeja apologies for the delay on this one!

Does this net relate to the Mesopelagic Trawl described here?
https://www.flotteoceanographique.fr/en/Facilities/Tooling/Sampling-by-Trawling/Plankton-Sampling-Tools

I'm just confused as to how the dimensions mentioned above could create a 50 m^2 opening?

JoBeja, 2022-09-30:
Hi @danibodc I believe it might be as, in my communications with the data provider, they sent a photo and it's the same as the close up of the net on the page you mention. 
Did you receive the file I sent via email on the day I created this ticket? It went to the vocab group email. The dimensions I include in the ticket were taken from that document and also after a clarification from our data provider (not the person that built the net). I also have just checked and the person who could know about the nets left IFREMER in 2013, I wouldn't know who to contact there to see if more info is available. Sorry, not much help at all :/

danibodc, 2022-09-30:
Hi @JoBeja, thank you! Not to worry.

Yes we received the document but I am struggling to see reference to this:
Rectangular fixed frame at the top measuring 5.6m*1.4m(?)

I can see the diameter is 8.4 m, but how does this relate to a frame size of 5.6 m by 1.4 m ? 

In the link provided above, I have since noticed that it states the net is approximately 40 m in length as opposed to 12.6 m we are dealing with here, so this is possibly not the right one after all!

JoBeja, 2022-09-30:
Hi @danibodc I don't really understand the measures from that diagram, and the provider wasn't able to help either, so please don't take my comments as an absolute truth. I've asked some Biology colleagues and we puzzled. Can anyone else at BODC help?

danibodc, 2022-10-04:
Thanks @JoBeja, we have now loaded the term:

NETT0202	Ifremer mesopelagos net

It will be live on the NVS from 0730 UTC tomorrow morning.

danibodc, 2022-10-05:
https://vocab.nerc.ac.uk/collection/L22/current/NETT0202/

closed 2022-10-05 by danibodc