nvs-vocabs/L22 #142: Request to modify term L22::TOOL1426

rhudak2020, 2022-07-14:
I would like to request either a modification of TOOL1426 (Pro-Oceanus HGTD-Pro {Hurricane} dissolved gas sensor) to include the GTD-Pro in its description/title or to add a new instrument (Pro-Oceanus GTD-Pro). They are both dissolved gas sensors. The only difference I can see is that the Hurricane model uses a supported tubular interface for faster equilibration. 

I welcome any thoughts on how to approach this issue.

SLBlakeman, 2023-05-19:
@rhudak2020 We have decided to create a new term for the GTD-Pro, we have also updated the definition of TOOL1426 to make it clear that it refers to the HGTD. Thank you for highlighting this.

The new term is now available on the NVS:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1920/
TOOL1920 - Pro-Oceanus GTD-Pro dissolved gas sensor

JordanAthertonNOC, 2024-08-23:
Hi @rhudak2020 

This past week I've been fixing some legacy issues with the manufacturer portion/prefix of L22s as part of a standardisation exercise for the vocabulary. This has resulted in updates to 270 terms. Pro-Oceanus terms were part of this, as some were input as `Pro Oceanus`, thus escaping discoverability when searching for `Pro-Oceanus`. 

Surprisingly (/thankfully!) this has only resulted in the discovery of one duplicate term following the updates. 

Therefore just a heads up - we will shortly be deprecating TOOL1920 and replacing it with the older TOOL0679.

rhudak2020, 2024-08-29:
Great, thanks!!

closed 2024-09-02 by JordanAthertonNOC