nvs-vocabs/L22 #144: NTR: Raytheon Standard 30 MF Gyrocompass

alex-tate, 2022-07-24:
## Device make
Raytheon (actually Raytheon Anschütz but recent L22 entries have just used Raytheon)

## Device range
Standard

## Device model
30 MF

## Model version
110-700 (but this seems too specific for the L22 entry, it is really a part number)

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
N/A

## Qualification (if applicable)
N/A

## Instrument type
Gyrocompass

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
N/A

## Short name
raytheon_standard_30mf

## Documentation
https://www.raytheon-anschuetz.com/what-we-do/product-range/standard-30-mf-gyro-compass/

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
0000-0002-7880-3271

danibodc, 2022-08-04:
This term has now been loaded and will be live on the NVS from 07:30 UTC tomorrow morning.

TOOL1829 - Raytheon Standard MF 30 gyrocompass

danibodc, 2022-08-05:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1829/

closed 2022-08-05 by danibodc