nvs-vocabs/L22 #145: NTR: Safran BlueNaute Gyrocompass

alex-tate, 2022-07-24:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Safran

## Device range
N/A

## Device model
BlueNaute

## Model version
Versions inlcude Compass, Premium, Titanium, Platinum - they differ in heading precision. It is unclear which version we have onboard the SDA as it isn't mentioned on the instrument case and as part of the ship's DP system we have no way of remotely accessing it.

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
N/A

## Qualification (if applicable)
N/A

## Instrument type
Gyrocompass, Attitude and Heading Reference System

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
N/A

## Short name
safran_bluenaute

## Documentation
https://www.safran-group.com/products-services/bluenauter-gyrocompass-attitude-heading-reference-system-inertial-navigation-system-maritime

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
0000-0002-7880-3271

danibodc, 2022-08-04:
This term has now been loaded and will be live on the NVS from 07:30 UTC tomorrow morning.

TOOL1830 - Safran BlueNaute gyrocompass, attitude and heading reference system series

danibodc, 2022-08-05:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1830/

closed 2022-08-05 by danibodc