nvs-vocabs/L22 #146: NTR: Observator OMC-116M Ultrasonic Wind Sensor

alex-tate, 2022-07-25:
## Device make
Observator

## Device range
N/A

## Device model
OMC-116

## Model version
M (marine version, varies only in the mounting option from the land version [OMC-116]) 

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
N/A

## Qualification (if applicable)
Full Metal

## Instrument type
Anemometer

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
N/A

## Short name
observator_omc116m

## Documentation
https://observator.com/products/omc-116-full-metal-ultrasonic-wind-sensor/

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
0000-0002-7880-3271

danibodc, 2022-08-04:
This term has now been loaded and will be live on the NVS from 07:30 UTC tomorrow morning:

TOOL1833 - Observator OMC-116M ultrasonic wind sensor

danibodc, 2022-08-05:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1833/

closed 2022-08-05 by danibodc