nvs-vocabs/L22 #147: NTR: FT Technologies FT702LT V22 Wind Sensor

alex-tate, 2022-07-25:
## Device make
FT Technologies

## Device range
N/A

## Device model
FT702LT (strictly the FT702 part is the model number while the LT indicates a digital variant versus LT/D which is the analogue variant)

## Model version
V22

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
N/A

## Qualification (if applicable)
N/A

## Instrument type
Anemometer

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
N/A

## Short name
ft_technologies_ft702lt

## Documentation
https://fttechnologies.com/wp-content/uploads/2019/04/FT702LT-V22-Datasheet.pdf

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
0000-0002-7880-3271

SLBlakeman, 2022-08-12:
Hello @alex-tate,

Looking at the instrument you have requested, would you be happy with a term such as FT702LT series, that included the various options FT702LT-V22-FF, FT702LT-V22-PM, FT702LT/D-V22-FF, FT702LT/D-V22-PM? Or would you prefer a term specifically for the FT702LT (digital display)? 

Thanks in advance.

alex-tate, 2022-08-12:
@SLBlakeman - your proposal for an FT702LT series sounds fine.

SLBlakeman, 2022-08-15:
@alex-tate 

Thank you for responding so quickly. I have now created the L22 term for you 
TOOL1834 - FT Technologies FT702LT-V22 anemometer series 
It is now available on our website at:
http://vocab.nerc.ac.uk/collection/L22/current/TOOL1834/

closed 2022-08-23 by SLBlakeman