nvs-vocabs/L22 #148: NTR: Biral SWS-200 Visibility and Present Weather Sensor

alex-tate, 2022-07-25:
## Device make
Biral

## Device range
N/A

## Device model
SWS-200

## Model version
N/A

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
N/A

## Qualification (if applicable)
N/A

## Instrument type
Visibility sensor, Present Weather Sensor (from L05, 'atmospheric transparency quantifiers', possibly also 'precipitation gauges' and 'meteorological packages')

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
N/A

## Short name
biral_sws200

## Documentation
https://www.biral.com/product/sws-200-visibility-present-weather-sensor/#product-overview

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
0000-0002-7880-3271

danibodc, 2023-05-17:
This term has now been loaded to L22 and will be live on the NVS from tomorrow morning:

TOOL1919	Biral SWS-200 visibility and present weather sensor

danibodc, 2023-05-18:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1919/

closed 2023-05-18 by danibodc