nvs-vocabs/L22 #149: NTR: Biral SWS-100 Visibility Sensor

alex-tate, 2022-07-25:
## Device make
Biral

## Device range
N/A

## Device model
SWS-100

## Model version
N/A

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
N/A

## Qualification (if applicable)
N/A

## Instrument type
Visibility sensor (from L05, 'atmospheric transparency quantifiers', possibly also 'precipitation gauges' and 'meteorological packages')

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
N/A

## Short name
biral_sws100

## Documentation
https://www.biral.com/product/sws-100-visibility-sensor/#product-overview

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
0000-0002-7880-3271

danibodc, 2022-08-04:
This term has been loaded and will be live on the NVS from 07:30 UTC tomorrow morning:

TOOL1832 - Biral SWS-100 visibility sensor

danibodc, 2022-08-05:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1832/

closed 2022-08-05 by danibodc