nvs-vocabs/L22 #150: NTR: Kipp & Zonen SMP22 Pyranometer

alex-tate, 2022-07-25:
## Device make
Kipp & Zonen

## Device range
N/A

## Device model
SMP22

## Model version
-A (refers to the analogue output of the sensor being 4 to 20mA rather the '-V' version which has analgue output as 0 to 1 V)

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
N/A

## Qualification (if applicable)
N/A

## Instrument type
Pyranometer (L05 term would be radiometers)

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
N/A

## Short name
kipp_zonen_smp22

## Documentation
https://www.kippzonen.com/Product/360/SMP22-Pyranometer

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
0000-0002-7880-3271

danibodc, 2022-08-04:
This term has now been loaded and will be live on the NVS from 07:30 UTC tomorrow morning.

TOOL1831 - Kipp and Zonen SMP22 pyranometer

danibodc, 2022-08-05:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1831/

closed 2022-08-05 by danibodc