nvs-vocabs/L22 #152: NTR: Campbell Scientific (Goodrich) 0871LH1 Freezing-Rain Sensor

alex-tate, 2022-07-25:
## Device make
Campbell Scientific (although they quote Goodrich as the manufacturer)

## Device range
N/A

## Device model
0871LH1

## Model version
N/A

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
N/A

## Qualification (if applicable)
N/A

## Instrument type
Feezing-rain sensor

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
N/A

## Short name
campbell_0871lh1

## Documentation
https://www.campbellsci.com/0871lh1

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
0000-0002-7880-3271

danibodc, 2022-10-11:
Hi @alex-tate

This term has now been loaded to L22 and will be live on the NVS from 0730 UTC tomorrow morning:
TOOL1855 Campbell Scientific (Goodrich) 0871LH1 freezing rain sensor

danibodc, 2022-10-12:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1855/

closed 2022-10-12 by danibodc