nvs-vocabs/L22 #153: NTR: Thies Clima 5.4110 Laser Precipitation Monitor

alex-tate, 2022-07-25:
## Device make
Thies Clima

## Device range
N/A

## Device model
5.4110

## Model version
00.000 (the first two digits indicates that our model can't have additional sensors connected [10 can] and the second three digits indicate the type of power supply (24 V in our model)

## Model alternative name (if applicable)
N/A

## Modification (if applicable)
N/A

## Qualification (if applicable)
N/A

## Instrument type
Disdrometer or Laser Precipitation Monitor (L05: probably 'precipitation gauges')

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
N/A

## Short name
thiesclima_5_4110

## Documentation
https://www.thiesclima.com/en/Products/Precipitation-Electrical-devices/?art=774

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
0000-0002-7880-3271

danibodc, 2022-10-11:
Hi @alex-tate 

This term has now been loaded to L22 and will be live on the NVS from 0730 UTC tomorrow morning:
TOOL1856	Thies Clima 5.4110 laser precipitation monitor

danibodc, 2022-10-12:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1856/

closed 2022-10-12 by danibodc