nvs-vocabs/L22 #155: Request to change linking L05 terms for L22 sub-bottom profilers

alex-tate, 2022-07-26:
The SDA has an installed Kongsberg (Simrad) TOPAS PS018 sub-bottom profiler (L22:TOOL0859). This term is linked to the broader term L05:378 (sediment profile imagers) but this seems incorrect as it doesn't capture still images or video as described by L05:378. Instead it should probably be tagged as L05:367 (>2000 Hz top-bandwidth sub-bottom penetrator and mud profiler systems) or L05:394 (Pinger sub-bottom profilers) although these terms both appear to overlap with each other. 

There is a similar problem with other L22 sub-bottom profilers currently tagged with L05:378.

louatbodc, 2022-09-12:
Hi Alex,

Indeed you are correct. I think there has been confusion with instruments like SPI cameras which is more appropriate to term L05:378. SPI cameras physically insert into the sediment and take a photo for example. We will indeed revise these linkages with other sub-bottom profilers.

We also agree that the TOPAS PS018 is suited to L05:367 (>2000 Hz top-bandwidth sub-bottom penetrator and mud profiler systems) and more granular terms describing different kinds of SBPs that use frequencies >=2000 Hz (e.g. chirp, pinger etc.). However, upon further investigation we believe the TOPAS PS018 is perhaps better described as a parametric sub-bottom profiler rather than pinger sub-bottom profiler. Do you agree?

Currently 'parametric sub-bottom profilers' does not exist in L05 and if you agreed, we would be required to raise the request with the current governance panel of L05. and it would take a few days for the term to be approved.

alex-tate, 2022-09-29:
Yes, a new term should probably be created in L05 for a parametric sub-bottom profiler. At the same time, the pinger sub-bottom profiler entry (L05:394) should probably be modified to explain that it is a linear system (as opposed to the non-linear parametric systems).

louatbodc, 2022-10-20:
Below are proposals for updates to the SeaDataNet device categories collection L05 to help address this request. We propose a new granular L05 term for a type of sub-bottom profiler - 'parametric' sub-bottom profilers. We also propose to update an existing term for 'pinger' sub-bottom profilers to help distinguish between the two

Parametric sub-bottom profilers (New term)
Marine seismic sources that send sound pulses into the sub-sea floor sediments. The sound pulses bounce off the sea floor and subsequent buried sediment layers according to differences in their acoustic impedance (hardness) and in turn, their density. Parametric sub-bottom profilers are non-linear systems that transmit two different higher frequencies that interact during sound propagation to generate a resultant lower operating frequency with narrow bandwidth. They typically operate at frequencies greater than 2 kHz and penerate to several hundreds of metres.

394 - Pinger sub-bottom profilers (Updated term)
Marine seismic sources that send sound pulses into the sub-sea floor sediments. The sound pulses bounce off the sea floor and subsequent buried sediment layers according to differences in their acoustic impedance (hardness) and in turn, their density. Pingers are linear, single channel, high frequency systems that typically operate at frequencies greater than 2 kHz and penerate to several tens of metres.

louatbodc, 2023-02-07:
Term 394 (Pinger sub-bottom profilers) has been updated to reflect linear systems.

Term 396 (parametric sub-bottom profilers) has been created to reflect dual frequency systems)

Please note it may up to 24 hours for changes to appear on our web systems.

closed 2023-02-07 by louatbodc