nvs-vocabs/L22 #156: Request to change linking L05 term for Heitronics C15 thermometer (TOOL1526)

alex-tate, 2022-07-26:
Currently the Heitronics C15 radioation thermometer L22 term (TOOL1526) is linked to the broader L05:134 (water temperature sensor). While it is true that the model that we have (the C15.85 version) is installed to measure the skin temperature of the ocean it doesn't measure temperature directly as L05:134 describes. Instead it is an (infrared) radiometer and should really be linked either to L05:122 (radiometer) or L05:302 (water body temperature sensor) or both.

danibodc, 2022-10-12:
Hi @alex-tate 

Yes I agree, I have updated the mappings to remove L05:134 and include both L05:122 (radiometer) and L05:302 (water body temperature sensor).

The mappings should be live from tomorrow morning after the overnight refresh.

danibodc, 2022-10-13:
Mappings are now live on the NVS:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1526/

closed 2022-10-13 by danibodc