nvs-vocabs/L22 #158: Query about Kongsberg (Simrad) SU90 fish finding sonar (TOOL1796)

alex-tate, 2022-07-27:
We have an SU90 installed on the SDA. However, this system comes with a variety of hull units that differ in terms of how far the sensor head can be lowered below the hull (see https://www.kongsberg.com/maritime/products/commercial-fisheries/fisherysonar/su90/simrad-su90-hull-units/ ). 

Confusingly these hull units are known as SU92 (the default), SU93 and SU94 (the one we have) and the latter code is how the system is referred to onboard the SDA  (i.e. 'we have an SU94', not 'we have an SU90 with an SU94 hull unit). The question is whether these model variations (of the SU90) are worth individual L22 entries or whether a note can be added to the exisiting SU90 description saying that they exist to avoid confustion when I link to the SU90 term.

I am about to submit a new request for the Simrad SC90 which has the same issue - it has two possible hull units, the SC90 and the SC94, of which we have the latter.

JordanAthertonNOC, 2022-10-19:
Hi @alex-tate 

I've discussed this with the team and we've decided the best course of action is to update the description of the existing term (SU90) to clarify the different available hull mounting configurations as the actual sensor remains unchanged. I will also apply this approach when creating the new term for the SC90. 

Will leave this ticket open until the description for the SU90 has been updated online (hopefully by tomorrow).

JordanAthertonNOC, 2022-10-20:
SU90 updated online to clarify hull mounting configurations:

http://vocab.nerc.ac.uk/collection/L22/current/TOOL1796/

closed 2022-10-20 by JordanAthertonNOC