nvs-vocabs/L22 #159: NTR: Kongsberg (Simrad) SC90 fish finding sonar

alex-tate, 2022-07-27:
## Device make
Kongsberg (Simrad)

## Device range
N/A

## Device model
SC90

## Model version
We have an SC94 hull unit - for discussion on this see #158 

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
Fish-finder echosounders

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
n/a

## Short name
simrad_sc90 or simrad_sc94

## Documentation
https://www.kongsberg.com/maritime/products/commercial-fisheries/fisherysonar/sc90/

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
0000-0002-7880-3271

JordanAthertonNOC, 2022-10-19:
One new L22 term created, anticipated to go live by 07:00 tomorrow BST.

Kongsberg (Simrad) SC90 fish finding sonar (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1865/)

JordanAthertonNOC, 2022-10-20:
Term now live on the NVS.

closed 2022-10-20 by JordanAthertonNOC