nvs-vocabs/L22 #161: NTR: Airmar CS229

rhudak2020, 2022-07-29:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Airmar

## Device range
n/a

## Device model
CS229

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
single beam transducer

## Does the tool describe a series of instruments? 
no

## Reference (if applicable)
n/a

## Short name
CS229

## Documentation
https://geo-matching.com/transducers-acoustic/cs229-low-frequency-cast-steel-transducer
https://drive.google.com/file/d/1LLD8rxEa-xCNI_Tyjw6rr40ARlQLBCM5/view?pli=1

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
0000-0002-0017-8330

SLBlakeman, 2022-09-05:
Dear Rebecca Hudak (@rhudak2020),

Thank you for submitting this request. We do not generally create L22 codes for standalone transducers. Is this transducer consistently part of a bigger system, e.g. linked to an echosounder? Is that system or echosounder in our L22 vocabulary? 

Kind regards,
Samantha Blakeman

rhudak2020, 2022-09-06:
@SLBlakeman apologies, the Airmar CS229 is part of our Knudsen 3260.

SLBlakeman, 2022-09-14:
@rhudak2020 apologies for the late reply, I have discussed with colleagues and we have decided to not create an L22 term for the transducer. This is the link to the Knudsen 3260 in case you need it:
[Knudsen Chirp 3260](http://vocab.nerc.ac.uk/collection/L22/current/TOOL0890)

closed 2022-09-21 by SLBlakeman