nvs-vocabs/L22 #162: NTR: Teledyne Reson SVP-70

rhudak2020, 2022-07-29:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Teledyne Reson

## Device range
SVP

## Device model
70

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
Sound Speed Sensor (ssv)

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
n/a

## Short name
SVP-70

## Documentation
http://www.teledynemarine.com/reson-svp-70

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
0000-0002-0017-8330

SLBlakeman, 2022-09-05:
Dear Rebecca Hudak  (@rhudak2020),

I'm working on this new L22 term, but I have a few questions. The specifications read as though the instrument has a temperature and pressure sensors, however, it doesn't mention the accuracy of their measurements which is strange. Do you know if it does have temperature and pressure sensors and if the data from these can be output? Or possible have the Operators Manual?

Thank you for your help,

Samantha Blakeman

SLBlakeman, 2022-09-15:
@rhudak2020 I have understood the instrument to have an internal temperature and pressure sensor. I have created the new term:
TOOL1843 - Teledyne Reson SVP 70 sound velocity profiler

It should be live on the NVS tomorrow morning.

SLBlakeman, 2022-09-21:
This term is now live on the NVS.

closed 2022-09-21 by SLBlakeman