nvs-vocabs/L22 #169: NTR: Zeiss Axiovert 200

roswri, 2022-10-27:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make

Zeiss 

## Device range

Axiovert 

## Device model

200 

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a
[Zeiss_axiovert200.xlsx](https://github.com/nvs-vocabs/L22/files/9881009/Zeiss_axiovert200.xlsx)

## Qualification (if applicable)
inverted

## Instrument type
fluorescence microscope

## Does the tool describe a series of instruments? 

individual

## Reference (if applicable)
n/a

## Short name
Zeiss Axiovert 200

## Documentation

https://www.southampton.ac.uk/microscopy/equipment/zeiss-axiovert-200-inverted-fluorescence.page

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
[Enter your ORCID if desired]

danibodc, 2022-11-16:
This term has now been loaded to L22 and will be live on the NVS from 0730 UTC tomorrow morning:

TOOL1880	Zeiss Axiovert 200 inverted fluorescence microscope

danibodc, 2022-11-17:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1880/

closed 2022-11-17 by danibodc