nvs-vocabs/L22 #173: NTR: Guralp Maris Ocean Bottom Seismometer

nadiakreimer, 2022-11-04:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Guralp

## Device range
N/A

## Device model
Maris Ocean Bottom Seismometer

## Model version
N/A

## Model alternative name (if applicable)
None by the manufacturer. Depending on how the instrument is configured its physical 'Nameplate' will indicate 'Response: Velocity...' or 'Response: Acceleration...', although the distinction is not part of the model name.

## Modification (if applicable)
Customer can select either velocity or acceleration variants depending on the preferred output (configurable prior to shipping). I.e., the seismometer can measure velocity or acceleration, depening on user request at the time of manufacturing. 
Please inform if two different L22 requests should be made to accommodate the two variants. The manufacturer does not distinguish the variants with a different model name.

## Qualification (if applicable)
Ocean Bottom

## Instrument type
Seismometer

## Does the tool describe a series of instruments? 
No.

## Reference (if applicable)
N/A

## Short name
Maris Seismometer

## Documentation
* Manufacturer website: [https://www.guralp.com/products/ocean-bottom-seismometers]
* Datasheet: [https://www.guralp.com/documents/DAS-OBS-0004.pdf]
* Technical manual: [https://www.guralp.com/documents/html/MAN-OBS-0013/]

## Mapping to external terminologies (optional)
N/A

## ORCID (optional)
0000-0002-8162-2383

danibodc, 2022-11-22:
This term has now been loaded to L22 and will be live on the NVS from 0730 UTC tomorrow morning. We created a single term covering velocity and acceleration variants.

TOOL1882	Guralp Maris ocean bottom seismometer

danibodc, 2022-11-23:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1882/

closed 2022-11-23 by danibodc