nvs-vocabs/L22 #175: NTR: Apollo AS-D1 DIC and d13C-DIC Analyzer

whshannon, 2022-11-29:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Apollo

## Device range
AS-D1

## Device model
Model AS-D1 DIC-δ13C Analyzer

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
DIC-δ13C Analyzer

## Instrument type
DIC-δ13C Analyzer

## Does the tool describe a series of instruments? 

## Reference (if applicable)
https://apolloscitech.com/dicdelta.html

## Short name
Apollo AS-D1

## Documentation
https://apolloscitech.com/dicdelta.html

## Mapping to external terminologies (optional)
https://vocab.nerc.ac.uk/collection/B75/current/ORG01043/

## ORCID (optional)
0000-0002-9081-1004

SLBlakeman, 2023-04-03:
This tool has been created and is now available on the NVS.

TOOL1906	Apollo SciTech AS-D1 Dissolved Inorganic Carbon (DIC) and delta13C preparation system and analyser https://vocab.nerc.ac.uk/collection/L22/current/TOOL1906/

whshannon, 2023-04-04:
Thanks so much for this (and for all of the other recent additions!) I've updated our BCO-DMO records accordingly.

closed 2023-05-19 by JordanAthertonNOC