nvs-vocabs/L22 #179: NTR: ECO_FLBBPC

vturpin, 2023-01-23:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Wetlabs

## Device range
n/a

## Device model
ECO_FLUOROMETER

## Model version
N/A

## Model alternative name (if applicable)
ECO_FLBBCP

## Modification (if applicable)

## Qualification (if applicable)

## Instrument type
FLUOROMETER_PHYCOCYANIN

## Does the tool describe a series of instruments? 
[Does the term describe a manufacturing 'series' of instruments, a 'system' of instruments, an 'individual' instrument or other?]

## Reference (if applicable)
https://www.seabird.com/eco-fluorometer/product-details?id=60429374754

## Short name
ECO_FLBBCP

## Documentation
https://www.seabird.com/eco-fluorometer/product?id=60429374754

## Mapping to external terminologies (optional)

## ORCID (optional)
[Enter your ORCID if desired]

JordanAthertonNOC, 2023-02-03:
Dear @vturpin 

Thank you for your request. Could I confirm whether this is actually a **WETLabs ECO Puck Triplet FLBBPC scattering fluorescence sensor**? Configured to measure optical scattering at 470, 532 or 650 nm, phycocyanin, and one of chl-a, CDOM, phycoerythrin, uranine or rhodamine? i.e. 2 fluorescence and 1 scattering measurements. 

https://www.seabird.com/asset-get.download.jsa?id=55352270602

Best

Jordan

JordanAthertonNOC, 2023-03-06:
Dear @vturpin 

Just following up on this to ensure you've seen the response.

Best

Jordan

vturpin, 2023-03-06:
Hi Jordan,
Thanks for reminder, I actually missed the previous message. 
The sensor is configure to measure chla, phycocyanin and optical backscatter.

However, it is hard to answer your question. 
From this : https://www.seabird.com/eco-fluorometer/product-details?id=60429374754 I understand that phycocyanin can be both fluorescence measurements or optical backscatter....

I will ask the PI. Louise Biddle for more precision.

vturpin, 2023-03-07:
I copy paste the answer of Louise here.
"
Hi Victor,
I agree with you, that it is a fluorescence measurement, but looking for the blue-green algae.
Cheers,
Louise
"
Hope that helps.
V.

JordanAthertonNOC, 2023-03-07:
Thanks for this @vturpin 

The last thing I need to confirm is whether this is the Puck + Triplet version of this ECO sensor (as there doesn't seem to be any information specifically for a `WET Labs ECO FLBBPC`). It looks like this is best signposted by the application (platform) and whether the output is in real time. From Sea-Bird: `The Puck is a miniature version of the popular ECO series of sensors, specifically designed for use in AUVs, profiling floats, and Slocum gliders with a dry science bay, where space is limited and power requirements are critical.`

Do you therefore know if this sensor was applied to an AUV / profiling float / Slocum glider and outputs in real time? 

We do have similar existing codes: 
https://vocab.nerc.ac.uk/search_nvs/L22/?searchstr=Puck%20Triplet&options=identifier,preflabel,altlabel,definition

vturpin, 2023-03-07:
Yes, this sensor is integrated to a glider (seaexplorer) and transmit data in real time.
This sensor is part of the same family as the one listed in you exemplar above.

JordanAthertonNOC, 2023-03-08:
One new L22 term created, this will be live on the NVS after ~08:30 CET tomorrow. 
WET Labs {Sea-Bird WETLabs} ECO Puck Triplet FLBBPC scattering fluorescence sensor (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1904/)

closed 2023-03-08 by JordanAthertonNOC