nvs-vocabs/L22 #180: NTR:

geo-sar, 2023-01-30:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Petersen

## Device range
n/a

## Device model
n/a

## Model version
n/a

## Model alternative name (if applicable)
[Alternative name for the model. E.g. Glider Payload CTD alternative name is GPCTD]

## Modification (if applicable)
[Describes any modifications to the instrument design, such as a manufactured device that has been modified 'in-house' or a plakton net that has been modified since its first description.]

## Qualification (if applicable)
[Qualification of the instrument type i.e. a word or words that further describe the instrument type (e.g. 'portable','automated','pressure-compensated','large' etc.). Sometimes these are used in manufacturer names.]

## Instrument type
trawl

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)
[Citation for the instrument (e.g. Hall and Bumpus 1937, 1944). Use if the instrument is manufactuered and built according to a design in a paper.]

## Short name
Petersen trawl

## Documentation
https://www.researchgate.net/publication/226192910_The_First_Russian_Fisheries_Research_Expedition_Centenary_of_the_Expedition_for_Fisheries_Research_along_the_Coast_of_Murman

## Mapping to external terminologies (optional)
[Is your submission related to any existing terms in external resources: please enter one or a list of URIs]

## ORCID (optional)
https://orcid.org/0000-0002-1700-3000

geo-sar, 2023-02-08:
I understand that there is no description for this instrument and it is a very specific instrument regarding a historical dataset. After consulting with other colleagues I believe that there is no need in creating a new term for this one...

JordanAthertonNOC, 2023-02-08:
Hi @geo-sar 

Thanks for this. If you do decide you would like a new L22 term for the Petersen Trawl in future please feel free to re-open this ticket.

closed 2023-02-08 by JordanAthertonNOC