nvs-vocabs/L22 #182: NTR:

kerfoot, 2023-02-08:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
[This is the name of the manufacturer or institution which made the instrument. If there is a manufacturer but you have not been able to confirm this information or there is no official manufacturer, input 'Inapplicable'. If you are creating a generic instrument (e.g. a generic temperature sensor, then input 'Unspecified'.]

Sea-bird Scientific http://vocab.nerc.ac.uk/collection/B75/current/ORG00226/
WET Labs http://vocab.nerc.ac.uk/collection/L35/current/MAN0026/

## Device range
[The manufacturer's instrument model range (e.g. WindObserver). If there is no official model range then input 'n/a'.]

ECO BB3SLO

## Device model
[Name of the instrument model. If there is no official model name, input 'n/a'.]

All backscattering version (blue 470nm, green 532nm, red 650nm or 660nm) of the ECO Triplet https://www.seabird.com/eco-triplet/product?id=60762467720
The calibration sheets refer to the instrument as a bb3slo instrument.  I will email these on request.

## Model version
[Name of the sub-model (e.g. 'D', 'v3.2'). If there is no official version/sub-model then input 'n/a'.]

bb3slo

## Model alternative name (if applicable)
[Alternative name for the model. E.g. Glider Payload CTD alternative name is GPCTD]

ECO BB3 Backscattering Meter

## Modification (if applicable)
[Describes any modifications to the instrument design, such as a manufactured device that has been modified 'in-house' or a plakton net that has been modified since its first description.]

## Qualification (if applicable)
[Qualification of the instrument type i.e. a word or words that further describe the instrument type (e.g. 'portable','automated','pressure-compensated','large' etc.). Sometimes these are used in manufacturer names.]

## Instrument type
[Please indicate what type of instrument it is (e.g. global positioning system, fluorometer, etc.)]

3 wavelength backscattering meter

## Does the tool describe a series of instruments? 
[Does the term describe a manufacturing 'series' of instruments, a 'system' of instruments, an 'individual' instrument or other?]

individual instrument

## Reference (if applicable)
[Citation for the instrument (e.g. Hall and Bumpus 1937, 1944). Use if the instrument is manufactuered and built according to a design in a paper.]

https://www.seabird.com/eco-triplet/product?id=60762467720

## Short name
[Shortened name of the instrument. This is usually a shortened version of name limited to make, model and version. It can also be an alternative name.]

bb3slo

## Documentation
[Please enter a link to the documentation describing the instrument. This could be a journal reference, manufacturer data sheet, web site, manual etc.]

https://www.seabird.com/asset-get.download.jsa?id=54894981701

## Mapping to external terminologies (optional)
[Is your submission related to any existing terms in external resources: please enter one or a list of URIs]

## ORCID (optional)
[Enter your ORCID if desired]

https://orcid.org/0000-0003-1405-4248

danibodc, 2023-04-26:
This term has been loaded to L22 and will be live on the NVS from tomorrow morning:
TOOL1911 WET Labs {Sea-Bird WETLabs} ECO Triplet BB3SLO scattering sensor

danibodc, 2023-04-27:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL1911/

closed 2023-04-27 by danibodc