nvs-vocabs/L22 #194: NTR: Turo Quoll XBT System

rhudak2020, 2023-09-13:
## Device make
Turo

## Device range
n/a

## Device model
Quoll 

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
XBT system

## Does the tool describe a series of instruments? 
n/a

## Reference (if applicable)
n/a

## Short name
Turo Quoll

## Documentation
https://www.turo.com.au/index.php/products/quoll-xbt

## Mapping to external terminologies (optional)

## ORCID (optional)
0000-0002-0017-8330

JordanAthertonNOC, 2023-09-14:
Hi Rebecca

Just to clarify - for this request are you looking for a term covering the system including the recorder (from the link provided) along with an XBT probe (description specific to Sippican XBT probes?), or just the recorder?

rhudak2020, 2023-09-14:
@JordanAthertonNOC I think just the recorder/system is sufficient

JordanAthertonNOC, 2023-09-18:
One new L22 has been created and will be live on the NVS from around ~06:00 UTC tomorrow. 

Turo Quoll XBT recorder (http://vocab.nerc.ac.uk/collection/L22/current/TOOL1988/)

closed 2023-09-18 by JordanAthertonNOC