nvs-vocabs/L22 #196: NTR: MC-800 Multi Corer

rhudak2020, 2023-11-10:
## Device make
Ocean Instruments

## Device range
MC series

## Device model
n/a

## Model version
MC-800

## Instrument type
Mulit Corer

## Does the tool describe a series of instruments? 
yes, MC series

## Reference (if applicable)
n/a

## Short name
MC-800

## Documentation
https://oceaninstruments.com/product/mc-800-multi-corer/

## Mapping to external terminologies (optional)
[Is your submission related to any existing terms in external resources: please enter one or a list of URIs]

## ORCID (optional)
0000-0002-0017-8330

roswri, 2024-01-31:
This term has now been created and is queued for publication on the NVS:
TOOL2008 - Ocean Instruments MC-800 Multi corer

roswri, 2024-02-01:
Term published:
https://vocab.nerc.ac.uk/collection/L22/current/TOOL2008/

closed 2024-02-01 by roswri