nvs-vocabs/L22 #197: NTR: MC-400 Multi Corer

rhudak2020, 2023-11-10:
## Device make
Ocean Instruments

## Device range
MC series

## Device model
MC-400

## Model version
n/a

## Model alternative name (if applicable)
n/a

## Modification (if applicable)
n/a

## Qualification (if applicable)
n/a

## Instrument type
Multi corer

## Does the tool describe a series of instruments? 
yes, mc series

## Reference (if applicable)
n/a

## Short name
MC-400

## Documentation
https://oceaninstruments.com/product/mc-400-multi-corer/

## Mapping to external terminologies (optional)
n/a

## ORCID (optional)
0000-0002-0017-8330

roswri, 2024-01-31:
This term has now been created and is queued for publication on the NVS:
TOOL2007 - Ocean Instruments MC-400 {Hedrick/Marrs} Multi corer

roswri, 2024-02-01:
Term published: https://vocab.nerc.ac.uk/collection/L22/current/TOOL2007/

closed 2024-02-01 by roswri