nvs-vocabs/L22 #201: NTR: Unspecified piston corer

jamcas-bas, 2023-11-13:
Please provide the following information (all fields are mandatory unless specified otherwise)

## Device make
Unspecified

## Device range
n/a

## Device model
n/a

## Model version
n/a

## Model alternative name (if applicable)

## Modification (if applicable)

## Qualification (if applicable)

## Instrument type
Piston corer

## Does the tool describe a series of instruments? 
individual

## Reference (if applicable)

## Short name
Unspecified piston corer

## Documentation
https://www.whoi.edu/what-we-do/explore/instruments/instruments-sensors-samplers/piston-corer/

## Mapping to external terminologies (optional)

## ORCID (optional)

vpaba, 2024-02-15:
Hi @jamcas-bas, thank you for your L22 submission. I am looking into this, and before initiating a new concept creation, I wondered whether any of these existing codes may meet your requirements:

- TOOL1169 - British Antarctic Survey (long) piston corer - BAS LG pist - Long Piston Corer (LPC) developed by BAS in collaboration with Marine Project Developments of Hull, originally trialled in 1992. Piston corers use heavy tubes which are plunged into the seafloor to extract samples of mud sediment. A piston inside the tube allows scientists to capture the longest possible samples. The corer can collect undisturbed marine sediment cores with barrel lengths of up to 30m.
- TOOL1178 - Unspecified trigger corer	 -  UnSpec Trigger - A trigger corer is a marine geological coring device used as part of the trigger system which allows a piston corer to free fall and collect cores. These trigger cores are often analysed by scientists.
- TOOL1176 - National Marine Facilities piston corer - NMF Piston - A marine piston corer part of the UK National Marine Equipment Pool (NMEP). The corer consists of a weight fitted with steel sample tubes. A PVC liner tube is fitted the full-length core barrel. A piston is fitted inside the PVC liner and is attached to a wire, which runs inside the corer barrel, through a hole in the centre of the weight and is attached to the main deployment wire. The corer is lowered on a wire to the seabed and a trigger system causes the core barrel to free fall into the sediment to collect samples. The action of the piston reduces internal friction allowing long cores to be collected. Core samples of 90mm and 110mm diameter can be taken. Up to five sections of 5.4 metre-long barrels can be fitted depending on the type of sediment being sampled.

If none of these fit, could you please provide some more information on the 'unspecified piston corer', and what sets it apart?

Thanks very much for your patience, and your help!

Kind regards,
Violetta

jamcas-bas, 2024-02-15:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Hello Violetta,

Thank you very much for looking over my submission.

My aim for the ‘unspecified piston corer’ code was to provide a concept to use when it is known that the core was collected using a piston corer (i.e. not a trigger corer, making TOOL1178 unsuitable) but exactly which model of piston corer is known (TOOL1169 and TOOL1176 both describe specific models of piston corer).

As an example, we have cores collected on cruise D172 (cruise report<https://www.bodc.ac.uk/resources/inventories/cruise_inventory/report/5081/>) which we know were collected using piston corers, though we do not know what model of piston corer.

I hope that this helps to explain the use case for this concept – please contact me if you have any further questions or suggestions?

Thanks,
James

This email and any attachments are intended solely for the use of the named recipients. If you are not the intended recipient you must not use, disclose, copy or distribute this email or any of its attachments and should notify the sender immediately and delete this email from your system. UK Research and Innovation (UKRI) has taken every reasonable precaution to minimise risk of this email or any attachments containing viruses or malware but the recipient should carry out its own virus and malware checks before opening the attachments. UKRI does not accept any liability for any losses or damages which the recipient may sustain due to presence of any viruses.

vpaba, 2024-02-20:
Hi @jamcas-bas, thank you for your reply and for the additional information and clarification.

I have created a new L22 concept with the following properties:

**ID** = TOOL2018
**PrefLabel** = Unspecified piston corer
**AltLabel** = UnSpec piston corer
**Description** = A generic term for a coring device designed to extract lengthy samples of undisturbed soft surface sediment. This type of corer is based on the original design by Swedish oceanographer Borje Kullenberg. Piston corers consist of a small gravity corer, a tripping arm with a weight, and a long, heavy corer containing a piston, connected to the ship by a wire. When the gravity corer touches the seafloor, the tripping mechanism is activated, thus allowing the piston corer to free fall to the bottom. Upon contact, the piston inside the corer stops at the sediment surface, thus creating a pressure differential that helps the soft material enter the corer's hollow open tube. The sediment sample is retained by a seal at the bottom of the device for retrieval. The sediment samples collected can be up to 30 m in length, and are less prone to compression and disturbance compared to samples from gravity corers.

Mappings between this L22 concept and L05::51, L35::MAN0001 and B75::UNSPEC were also created.

This concept should become visible on the NVS from tomorrow at: https://vocab.nerc.ac.uk/collection/L22/current/TOOL2018/

Please let me know if you spot any issues, and we'll reopen the ticket.

Kind regards,
Violetta

closed 2024-02-20 by vpaba

jamcas-bas, 2024-02-20:
Dear Violetta,

Thank you very much for your work on this concept.

I do have a suggestion for your consideration: as piston corers can also be used in coring lakes (e.g. TOOL2005<http://vocab.nerc.ac.uk/collection/L22/current/TOOL2005/> is a piston corer often used in shallower water such a lake) is it worth making the description more generic? Suggested changes to the description are in red below.

Description = A generic term for a coring device designed to extract lengthy samples of undisturbed soft surface sediment. This type of corer is based on the original design by Swedish oceanographer Borje Kullenberg. Piston corers consist of a small gravity corer, a tripping arm with a weight, and a long, heavy corer containing a piston, connected to the ship or platform by a wire. When the gravity corer touches the sediment surface, the tripping mechanism is activated, thus allowing the piston corer to free fall to the bottom. Upon contact, the piston inside the corer stops at the sediment surface, thus creating a pressure differential that helps the soft material enter the corer's hollow open tube. The sediment sample is retained by a seal at the bottom of the device for retrieval. The sediment samples collected can be up to 30 m in length, and are less prone to compression and disturbance compared to samples from gravity corers.

Thank you very much,
James

vpaba, 2024-02-21:
Dear @jamcas-bas,

Thank you for your quick response. Your suggestion makes sense, thanks for pointing this out! I have updated the description, which should be visible tomorrow (https://vocab.nerc.ac.uk/collection/L22/current/TOOL2018/).

Any other issues or comments please don't hesitate to get in touch.

Kind regards,
Violetta